esp-arduino-libs/ESP32_Display_Panel #150: Compilation of MIPI_DSI Example fails

ralphrmartin, 2025-01-09:
Library v 0.2.2

with ESP_Panel_Board_Supported.h set to say
` #define BOARD_ESP32_P4_FUNCTION_EV_BOARD
`
Compilation of MIPI_DSI Example fails as follows
```
In file included from /Users/ralph/Documents/Arduino/libraries/ESP32_Display_Panel/src/bus/base/esp_lcd_panel_io_3wire_spi.c:16:
/Users/ralph/Documents/Arduino/libraries/ESP32_Display_Panel/src/bus/base/esp_lcd_panel_io_additions.h:12:10: fatal error: base/esp_io_expander.h: No such file or directory
   12 | #include "base/esp_io_expander.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1
```

Lzw655, 2025-01-10:
Hi @ralphrmartin, 

Please reinstall the `ESP32_IO_Expander` library with the version `v0.1.0.` The latest `v1.0.0` is not compatible for this library now.

ralphrmartin, 2025-01-10:
Thanks, that now compiles. But it is unhelpful to release incompatible libraries...

Lzw655, 2025-01-10:
Yes, it's mainly for the next [release version v1.0.0](https://github.com/esp-arduino-libs/ESP32_Display_Panel/pull/137).

closed 2025-02-18 by Lzw655