esp-arduino-libs/ESP32_Display_Panel #154: building with esp-idf, error struct esp_lcd_rgb_panel_event_callbacks_t has no member named on_frame_buf_complete

alexf13e, 2025-01-31:
Hello,
I am trying to build using esp-idf and get the error in the title. A longer error is as follows:
```
.../Programming/esp32/remote_esp/managed_components/espressif__esp32_display_panel/src/lcd/ESP_PanelLcd.cpp: In member function 'bool ESP_PanelLcd::begin()':
.../Programming/esp32/remote_esp/managed_components/espressif__esp32_display_panel/src/lcd/ESP_PanelLcd.cpp:171:22: error: 'struct esp_lcd_rgb_panel_event_callbacks_t' has no member named 'on_frame_buf_complete'
  171 |         rgb_event_cb.on_frame_buf_complete = (esp_lcd_rgb_panel_frame_buf_complete_cb_t)onRefreshFinish;
      |                      ^~~~~~~~~~~~~~~~~~~~~
.../Programming/esp32/remote_esp/managed_components/espressif__esp32_display_panel/src/lcd/ESP_PanelLcd.cpp:171:47: error: 'esp_lcd_rgb_panel_frame_buf_complete_cb_t' was not declared in this scope; did you mean 'esp_lcd_rgb_panel_bounce_buf_fill_cb_t'?
  171 |         rgb_event_cb.on_frame_buf_complete = (esp_lcd_rgb_panel_frame_buf_complete_cb_t)onRefreshFinish;
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                               esp_lcd_rgb_panel_bounce_buf_fill_cb_t
[1095/1118] Building CXX object esp-idf/espressif__esp32_display_panel/CMakeFiles/__idf_espressif__esp32_display_panel.dir/src/lcd/ST77916.cpp.obj
ninja: build stopped: subcommand failed.
```

I'm not too confident in my use of the library. I have followed the how to use page and configured for my board in idf.py menuconfig as follows:
[image]
[image]

I set up the panel like this:
```
ESP_Panel* panel = new ESP_Panel();
const uint16_t display_width = panel->getLcdWidth();
const uint16_t display_height = panel->getLcdHeight();
const esp_lcd_panel_handle_t panel_handle = panel->getLcd()->getHandle();
const esp_lcd_touch_handle_t touch_handle = panel->getTouch()->getHandle();

(dimensions and handles used later for setting up slint ui)
```
Are there further steps I need to take? I have tried with each combination of the two uncheck options, and tried uncommenting the relevant lines in ESP_Panel_Board_Supported.h.
I also see in the error that a file for ST77916 is being built, but the display on the board uses ST7262, is this normal (i.e. all displays get built)?

Thanks

Lzw655, 2025-02-05:
Hi @alexf13e,

Sorry for the late. This is likely because you are using an older version of ESP-IDF 5.4. Please update to the latest TAG `v5.4` or `release/v5.4`.

Lzw655, 2025-02-06:
Thank you for your feedback! I think we can try to port the Slint in the future. For now, if you can use the LVGL, it's easy to avoid screen tearing with RGB LCD.