esp-arduino-libs/ESP32_Display_Panel #155: PlatformIO Build error

Snoopy-HSS, 2025-02-02:
Took the PlatformIO Example and get link errors:

.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/ESP-LCD/src/main.cpp.o:(.literal._Z5setupv+0x20): undefined reference to `ESP_Panel::ESP_Panel()'
.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/ESP-LCD/src/main.cpp.o:(.literal._Z5setupv+0x24): undefined reference to `ESP_Panel::init()'
.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/ESP-LCD/src/main.cpp.o:(.literal._Z5setupv+0x28): undefined reference to `ESP_Panel::begin()'
.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/ESP-LCD/src/main.cpp.o:(.literal._Z5setupv+0x2c): undefined reference to `ESP_Panel::getLcd()'
.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/ESP-LCD/src/main.cpp.o:(.literal._Z5setupv+0x30): undefined reference to `ESP_Panel::getTouch()'
.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/ESP-LCD/src/main.cpp.o: in function `setup()':
\ESP-LCD/src/main.cpp:26: undefined reference to `ESP_Panel::ESP_Panel()'
.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: \ESP-LCD/src/main.cpp:27: undefined reference to `ESP_Panel::init()'
.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: \ESP-LCD/src/main.cpp:34: undefined reference to `ESP_Panel::begin()'
.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: \ESP-LCD/src/main.cpp:37: undefined reference to `ESP_Panel::getLcd()'
.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: \ESP-LCD/src/main.cpp:37: undefined reference to `ESP_Panel::getTouch()'

Did I miss something?

Snoopy-HSS, 2025-02-03:
It's in VSCODE on Windows 11
Alle current Versions.

Lzw655, 2025-02-05:
Hi @Snoopy-HSS,

Sorry for the late. Please refer to the [documentation](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/docs/How_To_Use.md#configuration-instructions-1) and configure the corresponding files based on your development board type before using the `ESP_Panel` drivers.

Snoopy-HSS, 2025-02-06:
[image]

Did all the config.
But getting this undefined reference error...

Lzw655, 2025-02-08:
You are probably not using the default example from this repository, right? Based on the error log you provided, the example includes a `main.cpp` file, but the default example in this repository does not have it. I suggest trying the default example from this repository first for testing.

Snoopy-HSS, 2025-02-08:
[image]

[image]

Same if I user app.cpp instead of main.cpp

Linking .pio\build\ESP-LCD\firmware.elf
/.platformio/packages/toolchain-xtensa-esp-elf@13.2.0+20240530/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: /.platformio/packages/toolchain-xtensa-esp-elf@13.2.0+20240530/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/lib/esp32s3/no-rtti\libc.a(libc_a-getentropyr.o):(.literal+0x4): warning: _getentropy is not implemented and will always fail
/.platformio/packages/toolchain-xtensa-esp-elf@13.2.0+20240530/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: /.platformio/packages/framework-arduinoespressif32-libs/esp32s3/qio_opi\libesp_system.a(startup.c.obj):(.literal.__esp_system_init_fn_init_components0+0x8): undefined reference to `_Unwind_SetEnableExceptionFdeSorting'
/.platformio/packages/toolchain-xtensa-esp-elf@13.2.0+20240530/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: /.platformio/packages/framework-arduinoespressif32-libs/esp32s3/qio_opi\libesp_system.a(startup.c.obj): in function `__esp_system_init_fn_init_components0':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/startup.c:497:(.text.__esp_system_init_fn_init_components0+0x1e): undefined reference to `_Unwind_SetEnableExceptionFdeSorting'
/.platformio/packages/toolchain-xtensa-esp-elf@13.2.0+20240530/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: /.platformio/packages/framework-arduinoespressif32-libs/esp32s3/lib\libnewlib.a(reent_init.c.obj):(.iram1.0.literal+0x4): undefined reference to `_cleanup_r'
/.platformio/packages/toolchain-xtensa-esp-elf@13.2.0+20240530/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: /.platformio/packages/framework-arduinoespressif32-libs/esp32s3/lib\libnewlib.a(newlib_init.c.obj):(.data.s_stub_table+0x98): undefined reference to `_cleanup_r'
/.platformio/packages/toolchain-xtensa-esp-elf@13.2.0+20240530/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/ESP-LCD/src/app.cpp.o:(.literal._Z5setupv+0x24): undefined reference to `ESP_Panel::ESP_Panel()'
/.platformio/packages/toolchain-xtensa-esp-elf@13.2.0+20240530/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/ESP-LCD/src/app.cpp.o:(.literal._Z5setupv+0x28): undefined reference to `ESP_Panel::init()'
/.platformio/packages/toolchain-xtensa-esp-elf@13.2.0+20240530/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/ESP-LCD/src/app.cpp.o:(.literal._Z5setupv+0x2c): undefined reference to `ESP_Panel::begin()'
/.platformio/packages/toolchain-xtensa-esp-elf@13.2.0+20240530/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/ESP-LCD/src/app.cpp.o:(.literal._Z5setupv+0x30): undefined reference to `ESP_Panel::getLcd()'
/.platformio/packages/toolchain-xtensa-esp-elf@13.2.0+20240530/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/ESP-LCD/src/app.cpp.o:(.literal._Z5setupv+0x34): undefined reference to `ESP_Panel::getTouch()'
/.platformio/packages/toolchain-xtensa-esp-elf@13.2.0+20240530/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/ESP-LCD/src/app.cpp.o: in function `setup()':        
\OneDrive\Elektronik\ESP-LCD/src/app.cpp:26:(.text._Z5setupv+0x6b): undefined reference to `ESP_Panel::ESP_Panel()'
/.platformio/packages/toolchain-xtensa-esp-elf@13.2.0+20240530/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: \OneDrive\Elektronik\ESP-LCD/src/app.cpp:27:(.text._Z5setupv+0x73): undefined reference to `ESP_Panel::init()'
/.platformio/packages/toolchain-xtensa-esp-elf@13.2.0+20240530/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: \OneDrive\Elektronik\ESP-LCD/src/app.cpp:34:(.text._Z5setupv+0x7b): undefined reference to `ESP_Panel::begin()'
/.platformio/packages/toolchain-xtensa-esp-elf@13.2.0+20240530/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: \OneDrive\Elektronik\ESP-LCD/src/app.cpp:37:(.text._Z5setupv+0x8e): undefined reference to `ESP_Panel::getLcd()'
/.platformio/packages/toolchain-xtensa-esp-elf@13.2.0+20240530/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: \OneDrive\Elektronik\ESP-LCD/src/app.cpp:37:(.text._Z5setupv+0x9c): undefined reference to `ESP_Panel::getTouch()'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\ESP-LCD\firmware.elf] Error 1

Snoopy-HSS, 2025-02-08:
with platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.11/platform-espressif32.zip  it works...

Lzw655, 2025-02-11:
Although I cannot reproduce your issue, the platform link you provided also works and supports Arduino v3.1.1. I will use this link in the new version. Thanks for sharing!

closed 2025-02-18 by Lzw655