esp-arduino-libs/ESP32_Display_Panel #157: ERROR: no matching function for call to 'ledc_timer_config_t::ledc_timer_config_t(<brace-enclosed initializer list>)'

SimoDolcio, 2025-02-08:
hi,
**1. development environment is as follows.**
Verbose mode can be enabled via `-v, --verbose` option
inizio dello script
copia dei file di configurazione
copio il file lv_conf.h nella cartella delle librerie locale
srclvglconf:  C:\Progetti\ESP32\IOTHome\configFile\lv_conf.h
destlvglconf: C:\Progetti\ESP32\IOTHome\.pio\libdeps\esp32s3box\lv_conf.h
lv_conf.h non copiato, file già presente e non modificato
fine dello script
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32s3box.html
PLATFORM: Espressif 32 (6.9.0) > Espressif ESP32-S3-Box
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b 
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 46 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Waveshare_ST7262_LVGL @ 0.1.0
|-- ESP32_Display_Panel @ 0.1.8+sha.3f6322b
|-- ESP32_IO_Expander @ 0.0.3+sha.e5378da
|-- lvgl @ 8.3.11
|-- ArduinoJson @ 7.3.0
|-- Adafruit_VL53L0X @ 1.2.4
|-- FS @ 2.0.0
|-- SD @ 2.0.0
|-- SPI @ 2.0.0
Building in release mode

**2. Build Error**

pio/libdeps/esp32s3box/ESP32_Display_Panel/src/backlight/ESP_PanelBacklight.cpp: In constructor 'ESP_PanelBacklight::ESP_PanelBacklight(int, bool, bool)':
.pio/libdeps/esp32s3box/ESP32_Display_Panel/src/backlight/ESP_PanelBacklight.cpp:18:92: error: no matching function for call to 'ledc_timer_config_t::ledc_timer_config_t(<brace-enclosed initializer list>)'
     _channel_config(ESP_PANEL_BACKLIGHT_LEDC_CHANNEL_CONFIG_DEFAULT(io_num, light_up_level))
                                                                                            ^
In file included from .pio/libdeps/esp32s3box/ESP32_Display_Panel/src/backlight/ESP_PanelBacklight.h:9,
                 from .pio/libdeps/esp32s3box/ESP32_Display_Panel/src/backlight/ESP_PanelBacklight.cpp:8:
C:/Users/simon/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note: candidate: 'ledc_timer_config_t::ledc_timer_config_t()'
 } ledc_timer_config_t;
   ^~~~~~~~~~~~~~~~~~~
C:/Users/simon/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note:   candidate expects 0 arguments, 1 provided
C:/Users/simon/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note: candidate: 'constexpr ledc_timer_config_t::ledc_timer_config_t(const ledc_timer_config_t&)'
C:/Users/simon/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'const ledc_timer_config_t&'
C:/Users/simon/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note: candidate: 'constexpr ledc_timer_config_t::ledc_timer_config_t(ledc_timer_config_t&&)'
C:/Users/simon/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'ledc_timer_config_t&&'
*** [.pio\build\esp32s3box\lib4c2\ESP32_Display_Panel\backlight\ESP_PanelBacklight.cpp.o] Error 1
==================================================================== [FAILED] Took 70.94 seconds ====================================================================

please gime me a help to resolve the problem

thanks

Lzw655, 2025-02-08:
Hi @SimoDolcio,

I think your project has the following two issues:  
1. The version of `framework-arduinoespressif32` you are using is relatively old. I recommend referring to the [example](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/examples/PlatformIO/platformio.ini) to use a version that matches this library.  
2. The versions of `ESP32_Display_Panel` and its dependencies are also outdated. Similarly, I recommend referring to the [example](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/examples/PlatformIO/platformio.ini) to use the latest versions.

SimoDolcio, 2025-02-08:
hi @Lzw655 
I followed what was reported on the links. 
I get the following error:

**Configuring toolchain packages from a remote source...
Tool Manager: Installing espressif/toolchain-riscv32-esp @ 13.2.0+20240530
UnknownPackageError: Could not find the package with 'espressif/toolchain-riscv32-esp @ 13.2.0+20240530' requirements for your system 'windows_amd64'**

I also tried this solution
"
**Just delete C:\Users\<user>\.platformio\platforms\espressif32* and create the project again. It will download the latest version.**
"

Lzw655, 2025-02-11:
Could you try to use the following platform configurations?

```
platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.11/platform-espressif32.zip
platform_packages =
    platformio/framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#3.1.1
; Use high performance libraries to improve the performance, and avoid esp32s3 RGB LCD screen drifting issue
    platformio/framework-arduinoespressif32-libs@https://dl.espressif.com/AE/esp-arduino-libs/esp32-3.1.1-h.zip
```

Lzw655, 2025-02-12:
Hi, I'm very sorry, but I can't figure out this issue either. I recommend upgrading to a newer version.

Lzw655, 2025-02-13:
You're welcome!