esp-arduino-libs/ESP32_Display_Panel #158: ILI9341 LCD Display with touch

minou65, 2025-02-09:
Hi all
 I am trying to use this library with the ILI9341 LCD Display. I can compile and upload the firmware, But I recive the following error
```cpp
E (6) ESP_PanelHost: addHostSPI(80): Attempt to add with a different configuration
E (6) ESP_Panel: init(425): Add host failed
E (9) ESP_PanelHost: [ESP_ERR_INVALID_STATE] del(187): SPI[1] free failed
E (15) ESP_PanelHost: Delete panel host failed
E (19) ESP_Panel: begin(489): Panel is not initialized

```

I have Enable the ESP_PANEL_USE_CUSTOM_BOARD and made the following config

```cpp
#define ESP_PANEL_USE_LCD           (1)     // 0/1

#define ESP_PANEL_LCD_NAME          ILI9341

 /* LCD resolution in pixels */
#define ESP_PANEL_LCD_WIDTH         (320)
#define ESP_PANEL_LCD_HEIGHT        (240)

#define ESP_PANEL_LCD_BUS_TYPE      (ESP_PANEL_BUS_TYPE_SPI)

#define ESP_PANEL_LCD_BUS_HOST_ID           (1)     // Typically set to 1
#define ESP_PANEL_LCD_SPI_IO_CS             (5)
#if !ESP_PANEL_LCD_BUS_SKIP_INIT_HOST
#define ESP_PANEL_LCD_SPI_IO_SCK            (18)
#define ESP_PANEL_LCD_SPI_IO_MOSI           (23)
#define ESP_PANEL_LCD_SPI_IO_MISO           (19)    // -1 if not used
#endif
#define ESP_PANEL_LCD_SPI_IO_DC             (4)
#define ESP_PANEL_LCD_SPI_MODE              (0)     // 0/1/2/3, typically set to 0
#define ESP_PANEL_LCD_SPI_CLK_HZ            (40 * 1000 * 1000)
   // Should be an integer divisor of 80M, typically set to 40M
#define ESP_PANEL_LCD_SPI_TRANS_QUEUE_SZ    (10)    // Typically set to 10
#define ESP_PANEL_LCD_SPI_CMD_BITS          (8)     // Typically set to 8
#define ESP_PANEL_LCD_SPI_PARAM_BITS        (8)     // Typically set to 8

#define ESP_PANEL_LCD_IO_RST          (22)      // IO num of RESET pin, set to -1 if not use

#define ESP_PANEL_USE_TOUCH         (1)         // 0/1

#define ESP_PANEL_TOUCH_NAME        XPT2046

#define ESP_PANEL_TOUCH_BUS_HOST_ID     (0)     // Typically set to 0
#define ESP_PANEL_TOUCH_I2C_ADDRESS     (0)     // Typically set to 0 to use the default address.
                                                // - For touchs with only one address, set to 0
                                                // - For touchs with multiple addresses, set to 0 or the address
                                                //   Like GT911, there are two addresses: 0x5D(default) and 0x14
#if !ESP_PANEL_TOUCH_BUS_SKIP_INIT_HOST
#define ESP_PANEL_TOUCH_I2C_CLK_HZ      (400 * 1000)
                                                // Typically set to 400K
#define ESP_PANEL_TOUCH_I2C_SCL_PULLUP  (1)     // 0/1
#define ESP_PANEL_TOUCH_I2C_SDA_PULLUP  (1)     // 0/1
#define ESP_PANEL_TOUCH_I2C_IO_SCL      (18)
#define ESP_PANEL_TOUCH_I2C_IO_SDA      (8)

#define ESP_PANEL_TOUCH_IO_INT          (27)        // IO num of INT pin, set to -1 if not use

```

then in setup I do the following
```cpp
    panel = new ESP_Panel();
    panel->init();
    panel->begin();

    lvgl_port_init(panel->getLcd(), panel->getTouch());

    /* Lock the mutex due to the LVGL APIs are not thread-safe */
    lvgl_port_lock(-1);

    /* Create screen objects */
    lv_obj_t* label = lv_label_create(lv_scr_act());
    lv_label_set_text(label, "Hello, LVGL!");
    lv_obj_align(label, LV_ALIGN_CENTER, 0, 0);
    Serial.println("Display setup complete");

```
Is thre something that I have missing? As I read the lib should support this display

minou65, 2025-02-09:
found it, after I have enabled Log
```cpp
#define ESP_PANEL_ENABLE_LOG                (1)         // 0/1
```

I saw this errors
```
D (493) ESP_Panel: Panel init start
D (494) ESP_Panel: Use LCD
D (494) ESP_Panel: Use SPI bus
D (494) ESP_Panel: Create LCD bus
D (496) ESP_PanelHost: Add host SPI[1]
D (500) ESP_Panel: Create LCD device
D (503) ESP_Panel: Use touch
D (505) ESP_Panel: Use SPI bus
D (508) ESP_PanelHost: Host SPI[1] is already exist
E (513) ESP_PanelHost: addHostSPI(80): Attempt to add with a different configuration
E (520) ESP_Panel: init(425): Add host failed
D (524) ILI9341_CPP: Destroyed
E (527) ESP_PanelHost: [ESP_ERR_INVALID_STATE] del(187): SPI[1] free failed
E (534) ESP_PanelHost: Delete panel host failed
D (538) ESP_PanelHost: Destroyed
E (541) ESP_Panel: begin(489): Panel is not initialized
```

then I modified the following line to 2
```cpp
#define ESP_PANEL_TOUCH_BUS_HOST_ID         (2)     // Typically set to 1
```

Lzw655, 2025-02-10:
Hi @minou65,

Thank you for sharing your debugging process!

closed 2025-09-21 by minou65