esp-arduino-libs/ESP32_Display_Panel #160: Boards JSON file

Snoopy-HSS, 2025-02-18:
Were can i find the json Files for Waveshare Boards.
In example there are only expressif...

Lzw655, 2025-02-18:
Hi @Snoopy-HSS,

I'm very sorry, but my efforts are focused on adapting and maintaining the PlatformIO board files for all Espressif development boards. 

For Waveshare development boards, you can refer to the guidance in the README to create them yourself. Or maybe @Y1hsiaochunnn @H-sw123 can provide support for these.

Y1hsiaochunnn, 2025-02-18:
Nice suggestion, but what version of Platformio are you using? Platformio suspended the maintenance, so if you want to use the library, I recommend using 
https://github.com/pioarduino/platform-espressif32

I won't run adaptation tests on older versions of Platformio until it's back in a maintenance state

Snoopy-HSS, 2025-02-18:
Ok, created my own.
Worked like a charm.
Performance is way better since 1.0.0