esp-arduino-libs/ESP32_Display_Panel #161: Broken library requirements in PackageMetadata

APF-MRoger, 2025-02-18:
I'm not an expert, so feel free to tell me I am doing something wrong.

I am trying to anchor the version of the library that I use in my project to the latest 0.2.x + latest commits before 1.0.0 before I can push fixes to support 1.0.0.

In platformIO I am using this semantic versioning to attempt to pull 0.2.2 : 

```lib_deps =
    https://github.com/esp-arduino-libs/ESP32_Display_Panel.git@0.2.2
```

Unfortunately this somehow fails : 

```
PackageException: Package version 1.0.0+sha.a5007cc doesn't satisfy requirements 0.2.2 based on PackageMetadata <type=library name=ESP32_Display_Panel version=1.0.0+sha.a5007cc spec={'owner': None, 'id': None, 'name': 'ESP32_Display_Panel', 'requirements': '0.2.2', 'uri': 'git+https://github.com/esp-arduino-libs/ESP32_Display_Panel.git'}
```

As I said I might be doing something wrong, but I think something might be broken in the library PackageMetadata and it breaks the semantic versioning calls ?

Lzw655, 2025-02-19:
Hi @APF-MRoger,

No worries, feel free to ask. I have also encountered this issue before. The version should be the same as the TAG label and the correct URL should be as follows:

```
    https://github.com/esp-arduino-libs/ESP32_Display_Panel.git@v0.2.2
```

APF-MRoger, 2025-02-19:
> Hi [@APF-MRoger](https://github.com/APF-MRoger),
> 
> No worries, feel free to ask. I have also encountered this issue before. The version should be the same as the TAG label and the correct URL should be as follows:
> 
> ```
>     https://github.com/esp-arduino-libs/ESP32_Display_Panel.git@v0.2.2
> ```

I tried that, unfortunately it seems to break the semantic versioning identification too...

`SemanticVersionError: Invalid simple block 'v0.2.2'`

Lzw655, 2025-02-19:
How about the following?

```c
    https://github.com/esp-arduino-libs/ESP32_Display_Panel.git#v0.2.2
```

APF-MRoger, 2025-02-19:
> How about the following?
> 
>     https://github.com/esp-arduino-libs/ESP32_Display_Panel.git#v0.2.2

Yes, that works for "locking it in" to the specific version, and functionality. I was trying to use the following syntax (which worked until v1.0.0 release was published) : 

`https://github.com/esp-arduino-libs/ESP32_Display_Panel.git@^0.2.2`

The idea was to get it to pull 0.2.2 and any backward-compatible commits, and usually it works on other libraries (for example lvgl, I can specific `@^9.2.0` and it will load 9.2.2 + some commits). I'm not entirely sure how they make it work and why it doesn't in this case... probably some magic sauce !

Lzw655, 2025-02-19:
I think PlatformIO probably cannot recognize version numbers with the prefix `v`.

APF-MRoger, 2025-02-19:
> I think PlatformIO probably cannot recognize version numbers with the prefix `v`.

Not when using the "loose" `@` semantic versioning notation. On the flip side I can also use the SHA of a commit for the "hard" locking using the `#` notation.