esp-arduino-libs/ESP32_Display_Panel #162: V1.0.0 Doesn't Compile...

Shadowtrance, 2025-02-18:
V1.0.0 doesn't compile on esp-idf (5.4) or Arduino (3.1.1) if buttons are set to 0.
Buttons being set to 0 fail seems to be fixed by wrapping the below line and function with the max button check in esp_panel_touch.cpp
```
#if (CONFIG_ESP_LCD_TOUCH_MAX_BUTTONS > 0)

#endif
```
```
line 308
    ESP_UTILS_CHECK_FALSE_RETURN(readRawDataButtons(buttons_num), false, "Read buttons failed");
```
```
line 601
bool Touch::readRawDataButtons(int buttons_num)
```
But i still fails after that with.
```
espressif__esp32_display_panel/libespressif__esp32_display_panel.a(esp_panel_board.cpp.obj):(.literal._ZN9esp_panel7drivers6BusI2CC5ERKNS1_6ConfigE[_ZN9esp_panel7drivers6BusI2CC5ERKNS1_6ConfigE]+0x8): undefined reference to `vtable for esp_panel::drivers::BusI2C'
```

Lzw655, 2025-02-19:
Hi @Shadowtrance,

Thank you for reporting this issue! It can be fixed by [commit](https://github.com/esp-arduino-libs/ESP32_Display_Panel/pull/163/commits/efee7e2fcc49f4452fb7cc22c4c217955d7c2f5a).

Shadowtrance, 2025-02-19:
Nice!
It builds properly now :)
The last part was my fault as i forgot to enable the i2c driver in menufonfig.

closed 2025-02-19 by Shadowtrance