esp-arduino-libs/ESP32_Display_Panel #165: Tearing on BOARD_JINGCAI_ESP32_4848S040C_I_Y_3

ZN0WS, 2025-02-21:
Hi to all, 

I found out that running on the BOARD_JINGCAI_ESP32_4848S040C_I_Y_3 board created significant tearing on a simple image on an empty screen if other actions like Serial.println() were taking place.

I was able to eliminate tearing completely by reducing the speed in line 85 of BOARD_JINGCAI_ESP32_4848S040C_I_Y_3.h:
#define ESP_PANEL_BOARD_LCD_RGB_CLK_HZ          (12 * 1000 * 1000)

After this it works perfect without any tearing.

Now the questions: 
1. Is this an issue, is the clock speed wrong in the library file?
2. If it is not an issue (This is according to specs) but for whatever reason this causes tearing that can't be solved by enabling CONFIG_LVGL_PORT_AVOID_TEARING_MODE > 0, Should we add the possibility to change this for some boards in the ESP_Panel_Board_Supported.h rather than having to go into the library files that will get updated with every new version of the library?

Thanks and great work on v1. Working really well so far!

Lzw655, 2025-02-21:
Hi @ZN0WS,

Thanks for your analysis. Based on the description, I believe this issue is the same as the [screen drift issue](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/docs/envs/use_with_arduino.md#solution-for-screen-drift-issue-when-using-esp32-s3-to-drive-rgb-lcd-in-arduino-ide). When driving an RGB LCD, the ESP32-S3 is prone to this problem, especially when operating Flash or WiFi. I recommend following the linked guide and using the `high_perf` version of the arduino-esp32 SDK.

ZN0WS, 2025-02-21:
Thanks @Lzw655. As you mention, the high performance version of the SDK does improve significantly eliminating 99.9% of the tear.

I will close the issue as there are 2 valid work arounds:

1. Reduce Clock speed
2. Use high-perf version of SDK as mentioned above.

Thanks for the great work!

closed 2025-02-21 by ZN0WS