esp-arduino-libs/ESP32_Display_Panel #167: PlatformIO thinks flash size is only 4Mb

extremerotary, 2025-02-24:
Good morning,
I have gone through all the platformIO instructions to add the waveshare 7" LCD screen into my example lvgl port project. My code compiles without error. However, platformIO isn't respecting the build flags in the JSON file. I then tried to do it in the platformio.ini file itself and it still assumes 4Mb flash. Is there some other setting in this library that may be overriding my build definitions?

Platformio.ini
`
[env:BOARD_WAVESHARE_ESP32_S3_TOUCH_LCD_7]
board = BOARD_WAVESHARE_ESP32_S3_TOUCH_LCD_7
framework = arduino
platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.11/platform-espressif32.zip
platform_packages =
    platformio/framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#3.1.1
    platformio/framework-arduinoespressif32-libs@https://dl.espressif.com/AE/esp-arduino-libs/esp32-3.1.1-h.zip
monitor_speed = 115200
lib_deps =
    https://github.com/esp-arduino-libs/ESP32_Display_Panel.git
    https://github.com/esp-arduino-libs/ESP32_IO_Expander.git#v1.1.0
    https://github.com/esp-arduino-libs/esp-lib-utils.git#v0.2.0
    https://github.com/lvgl/lvgl.git#v8.4.0

; Explicitly set the flash size and maximum size (still isn't working)
board_build.flash_size = 8MB
board_build.maximum_size = 8388608
board_build.mcu = esp32s3
board_build.f_flash = 80000000L
board_build.partitions = default_8MB.csv

build_flags =
    ${common.build_flags}
    ${rgb_mipi_lcd.build_flags}
    -DESP_PANEL_BOARD_DEFAULT_USE_SUPPORTED=1
    -DBOARD_WAVESHARE_ESP32_S3_TOUCH_LCD_7
    -DSECURE_BOOT
    -DBOARD_HAS_PSRAM
    -DARDUINO_USB_MODE=1
    -DARDUINO_FLASH_SIZE=8388608
`

JSON file for the board:
`
{
  "build": {
    "arduino":{
      "ldscript": "esp32s3_out.ld",
      "partitions": "default_8MB.csv",
      "memory_type": "qio_opi"
    },
    "core": "esp32",
    "extra_flags": [
      "-DBOARD_HAS_PSRAM",
      "-DARDUINO_USB_MODE=1",
      "-DARDUINO_USB_CDC_ON_BOOT=0",
      "-DARDUINO_ESP32S3_DEV"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "hwids": [
      [
        "0X303A",
        "0x1001"
      ]
    ],
    "mcu": "esp32s3",
    "variant": "esp32s3"
  },
  "connectivity": [
    "wifi",
    "bluetooth"
  ],
  "debug": {
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "Waveshare ESP32-S3 Touch LCD 7\"",
  "upload": {
    "flash_size": "8MB",
    "maximum_ram_size": 327680,
    "maximum_size": 8388608,
    "require_upload_port": true,
    "speed": 921600
  },
  "url": "https://www.adafruit.com/product/5290",
  "vendor": "Espressif"
}
`

I appreciate any guidance you can provide. Thanks!

closed 2025-02-24 by extremerotary

Lzw655, 2025-02-25:
Hi @extremerotary,

Has your issue been resolved?

extremerotary, 2025-02-25:
Yes, I apologize. I ended up creating a custom partitions table with app0 being 5Mb and was able to upload.

Lzw655, 2025-02-25:
No worries! Feel free to ask.