esp-arduino-libs/ESP32_Display_Panel #168: Initializing SD Card on Waveshare LCD Touch 7"

extremerotary, 2025-02-26:
Good morning,
I am facing an issue where I am unable to initialize the SD Card on the Waveshare LCD Touch 7". I realize this may be a little out-of-scope for the purposes of this library, but I am looking for some guidance as I believe the issue is either with conflicting PINs or something to do with how the IO Expander is initialized because I then need to access the IO Expander to initialize and utilize the SD Card. I am programming in PlatformIO. 
The Waveshare's example code below compiles in ArduinoIDE without issue and properly outputs the 8Gb SD Card I have inserted into the display. 

```
#include "waveshare_sd_card.h"

// Initial Setup
void setup(){
    Serial.begin(115200);

    Serial.println("Initialize IO expander");
    /* Initialize IO expander */
    ESP_IOExpander_CH422G *expander = new ESP_IOExpander_CH422G((i2c_port_t)I2C_MASTER_NUM, ESP_IO_EXPANDER_I2C_CH422G_ADDRESS, I2C_MASTER_SCL_IO, I2C_MASTER_SDA_IO);
    expander->init();
    expander->begin();

    Serial.println("Set the IO0-7 pin to output mode.");
    expander->enableAllIO_Output();
    expander->digitalWrite(TP_RST , HIGH);
    expander->digitalWrite(LCD_RST , HIGH);
    expander->digitalWrite(LCD_BL , HIGH);

    // Use extended GPIO for SD card
    expander->digitalWrite(SD_CS, LOW);

    // Turn off backlight
    expander->digitalWrite(LCD_BL, LOW);

    // When USB_SEL is HIGH, it enables FSUSB42UMX chip and gpio19, gpio20 wired CAN_TX CAN_RX, and then don't use USB Function 
    expander->digitalWrite(USB_SEL, LOW);

    // Initialize SPI
    SPI.setHwCs(false);
    SPI.begin(SD_CLK, SD_MISO, SD_MOSI, SD_SS);
    if (!SD.begin()) {
        Serial.println("Card Mount Failed"); // SD card mounting failed
        return;
    }
    uint8_t cardType = SD.cardType();

    if (cardType == CARD_NONE) {
        Serial.println("No SD card attached"); // No SD card connected
        return;
    }

    Serial.print("SD Card Type: "); // SD card type
    if (cardType == CARD_MMC) {
        Serial.println("MMC");
    } else if (cardType == CARD_SD) {
        Serial.println("SDSC");
    } else if (cardType == CARD_SDHC) {
        Serial.println("SDHC");
    } else {
        Serial.println("UNKNOWN"); // Unknown Type
    }

    uint64_t cardSize = SD.cardSize() / (1024 * 1024);
    Serial.printf("SD Card Size: %lluMB\n", cardSize); // SD card size

    // Testing file system functionality
    listDir(SD, "/", 0);
    createDir(SD, "/mydir");
    listDir(SD, "/", 0);
    removeDir(SD, "/mydir");
    listDir(SD, "/", 2);
    writeFile(SD, "/hello.txt", "Hello ");
    appendFile(SD, "/hello.txt", "World!\n");
    readFile(SD, "/hello.txt");
    deleteFile(SD, "/foo.txt");
    renameFile(SD, "/hello.txt", "/foo.txt");
    readFile(SD, "/foo.txt");
    testFileIO(SD, "/test.txt");
    Serial.printf("Total space: %lluMB\n", SD.totalBytes() / (1024 * 1024)); // Total space
    Serial.printf("Used space: %lluMB\n", SD.usedBytes() / (1024 * 1024)); // Used space
}

// Main Loop
void loop() {
    
}
```

In my PlatformIO project, I have attempted to port this code over in such a way that I can call SD_init(). However, when I do so, I either get system crashes (when trying to define a new IO Expander), or the LCD itself turns my UI blue (likely an issue on conflicting PINs or order of operations between the board and LCD init() functions and the SD_init() function).

Is there any way you help guide me on the proper way to initialize and utilize the SD Card on this board? For reference, my setup() function looks like this:

```
void setup()
{

  Serial.begin(115200);
  //createWifiKey();
  delay(100); // Rumors say it helps avoid sporadical crashes after wakeup from deep-sleep

  WiFi.mode(WIFI_STA);
  delay(50);

  Serial.print("MAC Address: ");
  Serial.println(WiFi.macAddress());
  myMac = WiFi.macAddress();
  if (esp_now_init() == ESP_OK)
  {
    esp_now_register_recv_cb(OnDataRecv);
    esp_now_register_send_cb(OnDataSent);
  }
  if (esp_now_set_pmk(esp_now_key) != ESP_OK)
  {
    Serial.println("Error setting ESP-NOW PMK");
  }
  // load UI
  Serial.println("Initializing board");
  Board *board = new Board();
  board->init();
  assert(board->begin());
  Serial.println("Initializing LVGL");
  lvgl_port_init(board->getLCD(), board->getTouch());

  Serial.println("Creating UI");
  ui_init();   // function defined in my ui.h file

initializeSdCard();
}
```

And here are the 2 helper functions for the SD Card:

```
bool SD_init(void) {

 Serial.println("Initialize IO expander");
    /* Initialize IO expander */
/*
    ESP_IOExpander_CH422G *expander = new ESP_IOExpander_CH422G((i2c_port_t)I2C_MASTER_NUM, ESP_IO_EXPANDER_I2C_CH422G_ADDRESS, I2C_MASTER_SCL_IO, I2C_MASTER_SDA_IO);
    expander->init();
    expander->begin();

    Serial.println("Set the IO0-7 pin to output mode.");
    expander->enableAllIO_Output();
    expander->digitalWrite(TP_RST , HIGH);
    expander->digitalWrite(LCD_RST , HIGH);
    expander->digitalWrite(LCD_BL , HIGH);

    // Use extended GPIO for SD card
    expander->digitalWrite(SD_CS, LOW);
 // Initialize SPI
    SPI.setHwCs(false);
    SPI.begin(SD_CLK, SD_MISO, SD_MOSI, SD_SS);
*/

/* The above code causes the board to go into a crash loop*/

  if (!SD.begin()) {
      Serial.println("Card Mount Failed"); // SD card mounting failed
      return false;
  }
  return true;
}
void initializeSdCard()
{
  sdCard = SD_init();   // global bool variable. Causes UI to be blue
  if (sdCard)
  {
    // Check the total space and used space on the SD card
    uint64_t totalBytes = SD.totalBytes();
    uint64_t usedBytes = SD.usedBytes();
    freePercentage = ((totalBytes - usedBytes) * 100) / totalBytes;

    Serial.printf("SD Card Total Space: %llu bytes\n", totalBytes);
    Serial.printf("SD Card Used Space: %llu bytes\n", usedBytes);
    Serial.printf("SD Card Free Space: %d%%\n", freePercentage);
  }
}
```

Any guidance you can provide is greatly appreciated. I see the expander being initialized in the esp_panel_board.cpp file, but I don't understand how to use that code to also initialize the SD Card. Thanks for your time.

Lzw655, 2025-02-27:
Hi @extremerotary,

我认为这个问题是由于 LCD (`board`) 和 SDCard 重复创建和操作了 CH422G 导致的，其实你可以直接从 `board` 中获取获取 CH422G 的句柄并给 SDCard 使用，参考如下代码：

```c
void setup()
{
...
  initializeSdCard(board->getIO_Expander()->getBase());
}

bool SD_init(esp_expander::Base *expander) {
    /* Don't need to initialize CH422G and operate LCD/Touch pins here */
    // ESP_IOExpander_CH422G *expander = new ESP_IOExpander_CH422G((i2c_port_t)I2C_MASTER_NUM, ESP_IO_EXPANDER_I2C_CH422G_ADDRESS, I2C_MASTER_SCL_IO, I2C_MASTER_SDA_IO);
    // expander->init();
    // expander->begin();
    //Serial.println("Set the IO0-7 pin to output mode.");
    //expander->enableAllIO_Output();
    //expander->digitalWrite(TP_RST , HIGH);
    //expander->digitalWrite(LCD_RST , HIGH);
    // expander->digitalWrite(LCD_BL , HIGH);

    // Use extended GPIO for SD card
    expander->digitalWrite(SD_CS, LOW);
    ...
}

void initializeSdCard(esp_expander::Base *expander)
{
  sdCard = SD_init(expander);   // global bool variable. Causes UI to be blue
  ...
}
```

extremerotary, 2025-02-28:
This is working as expected. Thank you very much for your assistance!

Lzw655, 2025-02-28:
Great! You are welcome.

Lzw655, 2025-02-28:
Can you take a picture of the LCD display when there is a problem? Please also provide the project source code if it's ok?

Lzw655, 2025-03-24:
Hi @extremerotary,

Could you try to use the following platform packages? I think it might be related to the [RGB screen drift issue](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/docs/envs/use_with_arduino.md#solution-for-screen-drift-issue-when-using-esp32-s3-to-drive-rgb-lcd-in-arduino-ide).

```c
platform_packages =
    platformio/framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#3.1.1
; Use high performance libraries to improve the performance, and avoid esp32s3 RGB LCD screen drifting issue
    platformio/framework-arduinoespressif32-libs@https://dl.espressif.com/AE/esp-arduino-libs/esp32-3.1.1-h.zip
```

extremerotary, 2025-03-24:
I updated my platformio.ini file to this:
```
[platformio]
default_envs =
        BOARD_WAVESHARE_ESP32_S3_TOUCH_LCD_7

[common]
build_flags =
; Arduino related:
    -DCORE_DEBUG_LEVEL=0    ; Set to `5` for full debug output, `0` for none
; LVGL related:
    -DLV_CONF_INCLUDE_SIMPLE
    -DLV_LVGL_H_INCLUDE_SIMPLE
; Include path:
    -I src
; Others:
    ; -DDISABLE_ALL_LIBRARY_WARNINGS    ; Disable all library warnings

[spi_qspi_lcd]
build_flags =
; LVGL related:
    -DLV_COLOR_16_SWAP=1

[rgb_mipi_lcd]
build_flags =
; LVGL related:
    -DLV_COLOR_16_SWAP=0

;
[env:BOARD_WAVESHARE_ESP32_S3_TOUCH_LCD_7]
board = BOARD_WAVESHARE_ESP32_S3_TOUCH_LCD_7
framework = arduino
;platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.11/platform-espressif32.zip
platform = espressif32@53.3.11
platform_packages =
    platformio/framework-arduinoespressif32@3.1.1
    platformio/framework-arduinoespressif32-libs@https://dl.espressif.com/AE/esp-arduino-libs/esp32-3.1.1-h.zip
monitor_speed = 115200
lib_deps =
    https://github.com/esp-arduino-libs/ESP32_Display_Panel.git
    https://github.com/esp-arduino-libs/ESP32_IO_Expander.git#v1.1.0
    https://github.com/esp-arduino-libs/esp-lib-utils.git#v0.2.0
    https://github.com/lvgl/lvgl.git#v8.4.0

; Explicitly set the flash size and maximum size
board_build.flash_size = 8MB
board_build.maximum_size = 8388608
board_build.mcu = esp32s3
board_build.f_flash = 80000000L
board_build.partitions = partitions.csv

build_flags =
    ${common.build_flags}
    ${rgb_mipi_lcd.build_flags}
    -DESP_PANEL_BOARD_DEFAULT_USE_SUPPORTED=1
    -DBOARD_WAVESHARE_ESP32_S3_TOUCH_LCD_7
    -DLV_CONF_PATH="C:\Users\extre\OneDrive\DialedN\OneGauge\DigitalDashV2\src\lv_conf.h"
    -DSECURE_BOOT
    -DBOARD_HAS_PSRAM
    -DARDUINO_USB_MODE=1
    -DARDUINO_FLASH_SIZE=8388608
```
I have the same behavior with the blueish hue. 

I also tried to increase my buffer size between board init and begin in my setup function, but also the same result:
```
board = new Board();
  board->init();

  auto bus = static_cast<esp_panel::drivers::BusRGB *>(board->getLCD()->getBus());
  bus->configRGB_BounceBufferSize(800 * 20);

  assert(board->begin());
```

Lzw655, 2025-03-24:
I'm so sorry, I haven't encountered this phenomenon yet. 

Hi @Y1hsiaochunnn, do you have any suggestions for this problem?

Y1hsiaochunnn, 2025-03-24:
It might seem like this can only be caused by repeated calls to RGB GPIO pins (which I have encountered), but from the schematic of waveshare, they shouldn't interfere. Is Platformio occupying the RGB signal pin when configuring SD? Or hidden IO effects?

My suggestion is to test the demo of esp-idf or Arduino ide is it the difference on the platform?

extremerotary, 2025-03-24:
@Y1hsiaochunnn 
I am unable to set the PSRAM option from the ArduinoIDE documentation. There is no "OPI-PSRAM"

[image]

When I set to disabled, it crashes over and over without anything displayed.

When I set it to enabled, it still crashes over and over, but I do get an image displayed momentarily, and it looks like the correct hue.

12:24:58.984 -> Initializing board
12:24:58.984 -> [I][LvPort][esp_panel_board.cpp:0066](init): Initializing board (Waveshare:ESP32_S3_TOUCH_LCD_7)
12:24:59.026 -> [I][LvPort][esp_panel_board.cpp:0235](init): Board initialize success
12:24:59.026 -> [I][LvPort][esp_panel_board.cpp:0253](begin): Beginning board (Waveshare:ESP32_S3_TOUCH_LCD_7)
12:24:59.026 -> [I][Expander][esp_io_expander_ch422g.c:0075](esp_io_expander_new_i2c_ch422g): version: 0.1.0
12:24:59.138 -> E (357) lcd_panel.rgb: lcd_rgb_panel_alloc_frame_buffers(164): no mem for frame buffer
12:24:59.138 -> E (358) lcd_panel.rgb: esp_lcd_new_rgb_panel(347): alloc frame buffers failed
12:24:59.138 -> [E][LvPort][esp_panel_lcd_st7262.cpp:0079](init): Create refresh panel failed [ESP_ERR_NO_MEM]
12:24:59.184 -> [E][LvPort][esp_panel_lcd.cpp:0404](begin): Init failed
12:24:59.184 -> [E][LvPort][esp_panel_board.cpp:0315](begin): LCD device begin failed
12:24:59.184 -> 
12:24:59.219 -> assert failed: void setup() C:\Users\extre\Documents\Arduino\waveshare7_touch_lcd_sdcard_test_arduino_ide\waveshare7_touch_lcd_sdcard_test_arduino_ide.ino:100 (board->begin())
12:24:59.219 -> 
12:24:59.219 -> Backtrace: 0x40376692:0x3fcabbd0 0x4037d631:0x3fcabbf0 0x403843c9:0x3fcabc10 0x4200259a:0x3fcabd40 0x42043c1a:0x3fcabd90 0x4037e252:0x3fcabdb0
12:24:59.219 -> 
12:24:59.219 -> 
12:24:59.219 -> 
12:24:59.219 -> 
12:24:59.219 -> ELF file SHA256: 019608a9d
12:24:59.219 -> 
12:24:59.609 -> Rebooting...

If you can help me get the ArduinoIDE to compile without crashing, I can verify better.

extremerotary, 2025-03-24:
Here is the error I get with PSRAM enabled:

15:47:36.005 -> ESP-ROM:esp32s3-20210327
15:47:36.005 -> Build:Mar 27 2021
15:47:36.005 -> rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
15:47:36.005 -> Saved PC:0x40376c0d
15:47:36.005 -> SPIWP:0xee
15:47:36.005 -> mode:DIO, clock div:1
15:47:36.005 -> load:0x3fce2820,len:0x1188
15:47:36.005 -> load:0x403c8700,len:0x4
15:47:36.038 -> load:0x403c8704,len:0xbf0
15:47:36.038 -> load:0x403cb700,len:0x30e4
15:47:36.038 -> entry 0x403c88ac
15:47:36.395 -> PSRAM detected and enabled.
15:47:36.491 -> MAC Address: D8:3B:DA:AC:33:50
15:47:36.491 -> Initializing board
15:47:36.491 -> [I][LvPort][esp_panel_board.cpp:0066](init): Initializing board (Waveshare:ESP32_S3_TOUCH_LCD_7)
15:47:36.491 -> [I][LvPort][esp_panel_board.cpp:0235](init): Board initialize success
15:47:36.491 -> [I][LvPort][esp_panel_board.cpp:0253](begin): Beginning board (Waveshare:ESP32_S3_TOUCH_LCD_7)
15:47:36.491 -> [I][Expander][esp_io_expander_ch422g.c:0075](esp_io_expander_new_i2c_ch422g): version: 0.1.0
15:47:36.945 -> [I][Panel][esp_lcd_touch_gt911.c:0067](esp_lcd_touch_new_i2c_gt911): version: 1.1.1
15:47:36.945 -> [W][Panel][esp_lcd_touch_gt911.c:0144](esp_lcd_touch_new_i2c_gt911): Unable to initialize the I2C address
15:47:36.977 -> [I][Panel][esp_lcd_touch_gt911.c:0401](touch_gt911_read_cfg): TouchPad_ID:0x39,0x31,0x31
15:47:36.977 -> [I][Panel][esp_lcd_touch_gt911.c:0402](touch_gt911_read_cfg): TouchPad_Config_Version:88
15:47:36.977 -> [I][LvPort][esp_panel_board.cpp:0463](begin): Board begin success
15:47:36.977 -> Initializing LVGL
15:47:36.977 -> [I][LvPort][lvgl_v8_port.cpp:0761](lvgl_port_init): Initializing LVGL display driver
15:47:37.009 -> Creating UI
15:47:37.105 -> [E][Expander][esp_io_expander_ch422g.c:0197](write_output_reg): write_output_reg(197): Write WR-OC reg failed
15:47:37.105 -> [E][Expander][esp_io_expander.c:0103](esp_io_expander_set_level): esp_io_expander_set_level(103): Write Output reg failed
15:47:37.105 -> [E][LvPort][esp_expander_base.cpp:0217](digitalWrite): Set level failed [ESP_ERR_TIMEOUT]

extremerotary, 2025-03-28:
Any thoughts on why my ArduinoIDE build is crashing, but the same code is compiling and running with PlatformIO?

Y1hsiaochunnn, 2025-03-28:
Hi @extremerotary 
Normally, you need to initialize the screen and then initialize the SD card. After initializing the screen, the SPI pin of the screen will be idle and you can use the SD card.

In addition, there is no OPI PSRAM on the arduino. This is because we submitted the board support to the Arduino-ESP32, so the PSRAM can be turned on normally

extremerotary, 2025-04-05:
@Y1hsiaochunnn  Here's a video of the behavior. When the ui loads, then I get the io expander. It seems getting the expander is what causes it to turn blue. Then in the video, you can see that when I load the logo from the SD card, the screen goes back to normal color, but with scan lines in it. Any thoughts what is causing this? I can't get the code to work in ArduinoIDE.

H-sw123, 2025-04-07:
Hi @extremerotary 
The Arduino-esp32 SD API has been updated. If the CS pin is not specified in SD.begin, the default SS will be CS. The default CS will be integrated into the RGB LCD pin, resulting in color changes.
[https://github.com/H-sw123/arduino-esp32/blob/master/libraries/SD/src/SD.h](url)

You can try adding a CS pin in SD.begin, which should solve the problem. The specific modification is as follows:
SD.begin(SD_SS)

extremerotary, 2025-04-07:
> Hi [@extremerotary](https://github.com/extremerotary) The Arduino-esp32 SD API has been updated. If the CS pin is not specified in SD.begin, the default SS will be CS. The default CS will be [...]

@H-sw123 , thank you, that did it.

closed 2025-04-07 by extremerotary

Kanzll, 2025-04-10:
Thanks @Lzw655,
I changed SD.begin(SD_SS), and coloring problem is gone, but I got these errors in my console:

```
[  2647][E][esp32-hal-gpio.c:113] __pinMode(): Invalid IO 255 selected
[  2654][E][esp32-hal-periman.c:180] perimanGetPinBus(): Invalid pin: 255
[  2660][E][esp32-hal-gpio.c:176] __digitalWrite(): IO 255 is not set as GPIO.
[  2667][E][esp32-hal-periman.c:166] perimanSetPinBusExtraType(): Invalid pin: 255
[  2676][E][esp32-hal-periman.c:180] perimanGetPinBus(): Invalid pin: 255
[  2682][E][esp32-hal-gpio.c:176] __digitalWrite(): IO 255 is not set as GPIO.
...
[  3895][E][esp32-hal-periman.c:180] perimanGetPinBus(): Invalid pin: 255
[  3901][E][esp32-hal-gpio.c:176] __digitalWrite(): IO 255 is not set as GPIO.

```
I have understood that problem is with pin SD_SS, it's defined on -1 in "waveshare_sd_card.h".
I don't know if I must define any free pin, or should I upgrade my libraries which I think are very obsolete. you have help me to compile my code with this board with a specific version of sdk, and since I'm sticked with it because I can't compile with another combinaison (Arduino-ESP32-SDK, ESP32_Display_Panel and ESP32_IO_Expander.git)
here is my PlaformIO.ini 
```
[env:ESP-LCD]
platform = espressif32
board = ESP-LCD
framework = arduino
#board_build.partitions = partitions.csv

platform_packages =
        platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.3
        platformio/framework-arduinoespressif32-libs @ https://github.com/esp-arduino-libs/arduino-esp32-sdk.git#high_perf/v3.0.3

        
upload_speed = 921600
monitor_speed = 115200

#build_flags =
#        -DARDUINO_USB_CDC_ON_BOOT=0
#        -DCORE_DEBUG_LEVEL=5

build_flags =
        -DBOARD_HAS_PSRAM
        -DLV_CONF_INCLUDE_SIMPLE
        -DDISABLE_ALL_LIBRARY_WARNINGS
        -DARDUINO_USB_CDC_ON_BOOT=1
        -DCORE_DEBUG_LEVEL=1
        -DLV_LVGL_H_INCLUDE_SIMPLE
        -I src

lib_deps =
        https://github.com/esp-arduino-libs/ESP32_Display_Panel.git#99b2103
        https://github.com/esp-arduino-libs/ESP32_IO_Expander.git#v0.1.0
        https://github.com/lvgl/lvgl.git#release/v8.3
```