esp-arduino-libs/ESP32_Display_Panel #171: Upgrading from 0.2.2 to 1.0.0 - RGB panel invalid arguments

dinisfc, 2025-03-14:
I'm using a Waveshare ESP32-S3-Touch-LCD-4. I was initially using the files provided by Waveshare which included this library with the custom board configurations already applied for this board. I recently upgraded to 1.0.0 and now I get:

> [I][LvPort][esp_panel_board.cpp:0066](init): Initializing board (Custom:Custom)
> [I][LvPort][esp_panel_board.cpp:0235](init): Board initialize success
> [I][LvPort][esp_panel_board.cpp:0253](begin): Beginning board (Custom:Custom)
> [I][Expander][esp_io_expander_tca9554.c:0060](esp_io_expander_new_i2c_tca9554): version: 1.0.1
> [I][Panel][esp_lcd_st7701.c:0026](esp_lcd_new_panel_st7701): version: 1.1.1
> [E][Panel][esp_lcd_st7701_rgb.c:0065](esp_lcd_new_panel_st7701_rgb): esp_lcd_new_panel_st7701_rgb(65): invalid arguments
> [E][LvPort][esp_panel_lcd_st7701.cpp:0066](init): Create refresh panel failed [ESP_ERR_INVALID_ARG]
> [E][LvPort][esp_panel_lcd.cpp:0404](begin): Init failed
> [E][LvPort][esp_panel_board.cpp:0315](begin): LCD device begin failed

I'm using this with an lvgl ui from squareline and I've since updated my project using the porting example from version 1.0.0. Is this caused by improper configuration, possibly from adapting the configuration from the older version? For now, all code on the .ino file related to panel is exactly as the example at _examples/arduino/gui/lvgl_v8/squareline_port/_.

Config files below:
[OLD_esp_panel_config.zip](https://github.com/user-attachments/files/19245548/OLD_esp_panel_config.zip)
[NEW_esp_panel_config.zip](https://github.com/user-attachments/files/19245549/NEW_esp_panel_config.zip)

Lzw655, 2025-03-17:
Hi @dinisfc,

I have checked the `esp_panel_board_custom_conf.h` configuration file you provided and believe that modifying the following two points can resolve this issue:

- `    #define ESP_PANEL_LCD_SPI_CMD_BITS          (1)     // Typically set to 1`
- `    #define ESP_PANEL_LCD_SPI_CMD_BITS          (1)     // Typically set to 1`

You were misled by previous erroneous comments, this issue has been fixed in [85714fd](https://github.com/esp-arduino-libs/ESP32_Display_Panel/commit/85714fd3657897af655b9b5020b03cc486768508#diff-75dbd6cfa89672ebfd5d16b3be7e3ed106759e4c755a6fffbc6f6fdf9251c031L165-L166).

dinisfc, 2025-03-17:
Setting ESP_PANEL_BOARD_LCD_RGB_USE_CONTROL_PANEL to 1, shows this when running:

> [I][LvPort][esp_panel_board.cpp:0066](init): Initializing board (Custom:Custom)
> [I][LvPort][esp_panel_board.cpp:0235](init): Board initialize success
> [I][LvPort][esp_panel_board.cpp:0253](begin): Beginning board (Custom:Custom)
> [I][Expander][esp_io_expander_tca9554.c:0060](esp_io_expander_new_i2c_tca9554): version: 1.0.1
> [E][Panel][esp_lcd_panel_io_3wire_spi.c:0081](esp_lcd_new_panel_io_3wire_spi): esp_lcd_new_panel_io_3wire_spi(81): Invalid LCD command bytes
> [E][LvPort][esp_panel_bus_rgb.cpp:0668](begin): create control panel failed [ESP_ERR_INVALID_ARG]
> [E][LvPort][esp_panel_lcd.cpp:1028](processDeviceOnInit): Bus begin failed
> [E][LvPort][esp_panel_lcd_st7701.cpp:0062](init): Process device on init failed
> [E][LvPort][esp_panel_lcd.cpp:0404](begin): Init failed
> [E][LvPort][esp_panel_board.cpp:0315](begin): LCD device begin failed

I changed nothing else on the config file.

dinisfc, 2025-03-17:
That fixed the error but now the screen drifts. 
I have two tasks running, one on each core, one for the lvgl and other to read a sensor and save it to a file. Whenever the sensor task runs, the screen randomly shifts to one side, the touch still respects the ui positions though. Changing the core in which each task runs does not change the outcome. I use the arduino-esp32 version 3.1.1 since I've had a similar problem before in #156 but without the ui staying on the wrong location of the screen.

Lzw655, 2025-03-18:
So you mean you can't solve this problem according to the method in #156? Could you please take a video of the screen drift?

dinisfc, 2025-03-18:
Video showing the screen after startup
[startup.zip](https://github.com/user-attachments/files/19316464/startup.zip)
Shortly after, the drift gets worse as the ui starts to move on both axis; showing that touch still respects the ui button
[touchtest.zip](https://github.com/user-attachments/files/19316540/touchtest.zip)
I'm still using the high performance sdk for the ESP32S3 for version 3.1.1, yes.

Lzw655, 2025-03-19:
Just using high-performance SDK is not enough, it needs to combine the `RGB bounce buffer` to solve the problem (refer to [FAQ](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/docs/envs/use_with_arduino.md#solution-for-screen-drift-issue-when-using-esp32-s3-to-drive-rgb-lcd-in-arduino-ide)), you can try the following two measures:

1. Increase the RGB bounce buffer. Set `#define ESP_PANEL_BOARD_LCD_RGB_BOUNCE_BUF_SIZE (ESP_PANEL_BOARD_WIDTH * 10)` in the *esp_panel_board_custom_conf h* file, and continue to increase to `ESP_PANEL_BOARD_WIDTH * 20` if the problem persists. Please note that when the anti-tear function is turned on in the *squareline_port* example, the `RGB bounce buffer` will be set by code by default. Please modify [here](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/7eb03b668ef6f3eb839c7b261c2f55ecb9adf546/examples/arduino/gui/lvgl_v8/simple_port/simple_port.ino#L39) directly to increase it.
2. Ensure that the lvgl task is on the same core as the task that executes `board- > begin ()`. For example, if `board- > begin ()` runs in `setup ()` by default, you can set the *lvgl_v8_port h* file `#define LVGL_PORT_TASK_CORE (ARDUINO_RUNNING_CORE)`

dinisfc, 2025-03-19:
Thanks, `#define ESP_PANEL_BOARD_LCD_RGB_BOUNCE_BUF_SIZE (ESP_PANEL_BOARD_WIDTH * 10)` fixed it.

closed 2025-03-19 by dinisfc