esp-arduino-libs/ESP32_Display_Panel #174: readRawData always returns true & setDisplayOnOff doesn't turn off the LCD

dinisfc, 2025-03-19:
I'm planning on having a display idle with the LCD panel off and have the touch panel always waiting for touch, once touch is detected the LCD comes back on. 

I'm trying to use `readRawData` to detect touch, regardless of location, strength or number of touch points. When using `readRawData`, it always returns `true`, even if there's no touch detected and for any value for the timeout, it returns instantly or almost instantly. 

At the same time, I'm trying to turn on and off the LCD using `setDisplayOnOff` but it doesn't seem to do anything, I've tried using `enable_on` as _bool_ and as an _int_ but the LCD never turns off. 

Before, using version 0.2.2, both `displayOn `and `displayOff` worked no problem, and `readRawData` worked too. 
I can also confirm that the program is not getting stuck on any of those functions, for any possible reason, as the program continues running.

I'm sure there's better ways of doing this, but using `readRawData` seems the simplest since I basically only need a toggle of there being touch or not.

Lzw655, 2025-03-20:
Hi @dinisfc,

I'm sorry you encountered this problem in the new version. Please refer to the [steps](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/docs/envs/use_with_arduino.md#screen-not-working-in-arduino-ide-how-to-debug) to open the debug logs and send me the complete text for analysis. Thank you!

dinisfc, 2025-03-20:
Using readPoints did work. 
I turned on the debug logs and I saw no reference to setDisplayOnOff, while readPoints and the functions it uses all showed up.

Lzw655, 2025-03-21:
Maybe `setDisplayOnOff ()` is not triggered? Otherwise, there should be a log showing that this function was called.

dinisfc, 2025-03-21:
The function that uses it is:
```
void scrIdle( void * pvParameters ){
  auto lcd = board->getLcd();
  auto touch = board->getTouch();
  vTaskDelay(10000/portTICK_PERIOD_MS);
  TouchPoint points[5];
  while(1){ 
    if(touch->readPoints(points,5,1000)>0){
      lcd->setDisplayOnOff(true);
      vTaskDelay(30000/portTICK_PERIOD_MS);
    }
    lcd->setDisplayOnOff(false);
    vTaskDelay(10/portTICK_PERIOD_MS);
  }
}
```
Which is a freeRTOS task initialiazed by `xTaskCreatePinnedToCore(scrIdle, "screenIdle", 4096, NULL, 1, &screenIdle, 1);`
`board` is initialized the same way as the example. 
On core 1 is LVGL and this task, on core 0 is a task that reads from a air quality sensor and saves it to storage and computes some values.

After trying different versions of the function above, the timeout parameter of readPoints doesn't seem to actually change anything, unless I messed up again. Whatever value is on the third parameter, the timeout always ends instantly, which eventually causes this errors multiple times:

> E (24803) lcd_panel.io.i2c: panel_io_i2c_tx_buffer(177): i2c transaction failed
> E (25806) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(136): i2c transaction failed
> [E][Panel][esp_lcd_touch_gt911.c:0229](esp_lcd_touch_gt911_read_data): esp_lcd_touch_gt911_read_data(229): I2C read error!
> [E][LvPort][esp_panel_touch.cpp:0302](readRawData): Read data failed [ESP_ERR_TIMEOUT]
> [E][LvPort][esp_panel_touch.cpp:0454](readPoints): Read raw data failed

And, eventually, a full crash and reboot. This can be avoided by using the `vTaskDelay` outside of the `if`, but the touch detection doesn't work if the task is stopped on that delay.

extremerotary, 2025-03-22:
@dinisfc , which LCD screen are you using? Is it one of the supported models? If so, check out the model.h file from ESP32_Display_Panel > src > board > supported > {make} > {model}.h
Then look for ESP_PANEL_BOARD_BACKLIGHT_TYPE. What kind is it?
For example, mine is the waveshare 7" touch. My backlight isn't PWM and is part of the expander, so I have to do:
```
auto backlight = board->getBacklight();
backlight->on();
```
Not sure if that'll help you, but that works for my screen.

Lzw655, 2025-03-24:
> The function that uses it is:
>
> ```
> void scrIdle( void * pvParameters ){
>   auto lcd = board->getLcd();
>   auto touch = board->getTouch();
>   vTaskDelay(10000/portTICK_PERIOD_MS);
>   TouchPoint [...]

Hi @dinisfc,

The [timeout](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/7eb03b668ef6f3eb839c7b261c2f55ecb9adf546/src/drivers/touch/esp_panel_touch.hpp#L407) parameter of `readPoints()` is only valid if the INT pin is used because we can only be notified when the Interrupt signal is triggered.

Lzw655, 2025-03-25:
Hi @extremerotary, The `configInterruptActiveLevel()` should be called between `board->init();` and `board->begin()`.

dinisfc, 2025-06-18:
I've since changed to the ESP32-S3-Touch-LCD-4.3B, also by Waveshare. It uses a different LCD driver, but, in general, I made it run the same code, yet, I still can't get setDisplayOnOff to work. This is on version 1.0.2. Doesn't give an error or anything, it simply doesn't do anything to the screen.

Lzw655, 2025-07-07:
Hi @dinisfc, 
Sorry for the wrong mention before. 

Regarding the issue that `setDisplayOnOff()` of ST7262 doesn't work, I rechecked the code. In fact, when ST7262 is configured with a valid `disp_gpio_num` (or `ESP_PANEL_BOARD_LCD_RGB_IO_DISP`), this function is supported. However, development boards including `ESP32-S3-Touch-LCD-4.3B` do not support this pin. Therefore, the driver will automatically cancel the support for `setDisplayOnOff()` in this case. Refer to the [code here](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/2128a6e8c2de7231626c32a5642b1efd8fdbf76d/src/drivers/lcd/esp_panel_lcd.cpp#L1076-L1089).