esp-arduino-libs/ESP32_Display_Panel #181: Cant build anymore with the v1.0.0

lucyd-dev, 2025-03-24:
Heya,
i saw that this libary had upgraded to 1.0.0, so i tryed to setup my existing Project with the new changes. 
I set everything up, like in the PlatformIO example. But i cant even download the needed lib
```
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\steve\\.platformio\\.cache\\tmp\\pkg-installing-x1jje62z\\arduino-esp32-libs-all-release_v5.3-cfea4f7c98\\esp32s2\\include\\espressif__esp_matter\\connectedhomeip\\connectedhomeip\\src\\app\\clusters\\concentration-measurement-server\\concentration-measurement-cluster-objects.h'
```
or when I try it with a local folder im getting 
```
C:/esp32-libs/esp32-3.1.1-h/arduino-esp32-libs-all-release_v5.3-cfea4f7c98/esp32s3/include/newlib/platform_include/sys/lock.h:9:10: fatal error: sdkconfig.h: No such file or directory
```

Everything I try doesnt work for ours and Im getting tired ^^" 
Could u help me?

```ini
[platformio]
default_envs = waveshare_esp32s3_touch_lcd_4_3

[env]
framework = arduino
; The `espressif32` latest official version does not support Arduino v3.1.x, temporarily using a third-party version
; platform = espressif32
platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.11/platform-espressif32.zip
platform_packages =
	platformio/framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#3.1.1
	; Use high performance libraries to improve the performance, and avoid esp32s3 RGB LCD screen drifting issue
    platformio/framework-arduinoespressif32-libs@symlink:///esp32-libs/esp32-3.1.1-h/arduino-esp32-libs-all-release_v5.3-cfea4f7c98
    ; platformio/framework-arduinoespressif32-libs@https://dl.espressif.com/AE/esp-arduino-libs/esp32-3.1.1-h.zip

monitor_speed = 115200
monitor_filters =
	esp32_exception_decoder
	colorize
upload_speed = 921600

lib_deps =
	https://github.com/esp-arduino-libs/ESP32_Display_Panel.git
	https://github.com/esp-arduino-libs/ESP32_IO_Expander.git#v1.1.0
    https://github.com/esp-arduino-libs/esp-lib-utils.git#v0.2.0
    https://github.com/lvgl/lvgl.git#v8.4.0
	adafruit/Adafruit MCP23017 Arduino Library@^2.3.2
	thomasfredericks/Bounce2@^2.72

[common]
build_flags =
	-D BOARD_HAS_PSRAM
	-D LV_CONF_INCLUDE_SIMPLE
	-D LV_LVGL_H_INCLUDE_SIMPLE
	; -D DISABLE_ALL_LIBRARY_WARNINGS
	-D CORE_DEBUG_LEVEL=5
	-D ARDUINO_USB_MODE=1
	-D ARDUINO_USB_CDC_ON_BOOT=1
	-I src
	-D TP_RST=1
	-D LCD_BL=2
	-D LCD_RST=3
	-D USB_SEL=5
	-D I2C_MASTER_NUM=I2C_NUM_0
	-D I2C_MASTER_SDA_IO=8
	-D I2C_MASTER_SCL_IO=9
	-D SD_MOSI=11
	-D SD_CLK=12
	-D SD_MISO=13
	-D SD_CS=4
	-D SD_SS=-1

[spi_qspi_lcd]
build_flags =
; LVGL related:
    -DLV_COLOR_16_SWAP=1

[rgb_mipi_lcd]
build_flags =
; LVGL related:
    -DLV_COLOR_16_SWAP=0

[env:waveshare_esp32s3_touch_lcd_4_3]
build_flags =
    ${common.build_flags}
; Add custom build flags here:
    ; ${rgb_mipi_lcd.build_flags}
    ; ${spi_qspi_lcd.build_flags}
board = waveshare_esp32s3_touch_lcd_4_3

```

Lzw655, 2025-03-25:
Hi @lucyd-dev,

I tried to reproduce this problem your way, but everything is fine. Sorry, I can't give you a better suggestion.

lucyd-dev, 2025-03-25:
Okay, thats weird. Even when I create an test project like your example I get the same errors. I also reinstalled PlatformIO.
The only thing that seems to be fixed is, that now both libs produces the same error "sdkconfig.h: No such file or directory".

Before I changed to 1.0, it worked seemlessly fine and the only thing i changed in platformio.ini where the platform/_packages as you suggested in the new example. Im confused

lucyd-dev, 2025-03-25:
Nvm, i found the issue. 
I changed memory_type to opi, as I missunderstood your guide (https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/docs/board/board_waveshare.md#arduino-ide). I reverted it and now it compiles kinda. I just need to fix my code haha
Im sorry that I bothered you, but thx that you tryed to help <3 you can close the issue.

Lzw655, 2025-03-25:
Thank you for your feedback! No worries, feel free to ask.

closed 2025-03-25 by Lzw655