esp-arduino-libs/ESP32_Display_Panel #182: XPT2046 Touchscreen Controller Conflict

joelccrouch, 2025-04-03:
I'm trying to read touch points in my Arduino sketch, and I occasionally run into SPI conflicts as shown below:

E (15088) spi_master: spi_device_acquire_bus(977): Cannot acquire bus when a polling transaction is in progress.
E (15088) lcd_panel.io.spi: panel_io_spi_rx_param(265): acquire spi bus failed
E (15094) xpt2046: xpt2046_read_register(189): XPT2046 read error!
E (15100) xpt2046: xpt2046_read_data(218): XPT2046 read error!
E (15105) ESP_PanelTouch: [ESP_ERR_INVALID_STATE] readRawData(183): Read data failed
E (15113) ESP_PanelTouch: readPoints(221): Read raw data failed

When this happens, the touchscreen controller stops working, and I have to reset board to fix it.  How can I gracefully get the x,y coordinates of touches without conflict?

Lzw655, 2025-04-07:
Hi @joelccrouch,

Since the LCD and touch use the same SPI bus, you need to use a semaphore to exclusively operate them to ensure that they are not used simultaneously.

joelccrouch, 2025-04-07:
Yes, I realized as much, but I don't know where in the library to put the mutex/semaphore. I found this resistive touchscreen calibration:
https://www.google.com/url?sa=t&source=web&rct=j&opi=89978449&url=https://github.com/jakpaul/lvgl_touch_calibration&ved=2ahUKEwik5cXqwsWMAxXdmYkEHWZBHoAQFnoECBUQAQ&usg=AOvVaw1DUje-TzRiyzqwh50sCgzm

I don't understand what I am supposed to put here:

void your_indev_read_cb(
        lv_indev_t *indev,
        lv_indev_data_t *indevData
    ) {
    
    indevData->state = /* Implement this */;
    indevData->point.x = /* Implement this */;
    indevData->point.y = /* Implement this */;
}

Isn't the corrected x,y points supposed to be assigned here?

Thanks for your help.

Lzw655, 2025-04-07:
Could you share the project? I can try to implement it for you.

Lzw655, 2025-04-09:
Hi @joelccrouch, 

You can refer to the modifications in the attachment to achieve mutually exclusive access between LCD and touch. BTW, you are using an old version of ESP32_Display_Panel. It is recommended to use the latest v1 version.

joelccrouch, 2025-04-09:
I replaced my file with your lvgl_port_v8.cpp, but I get the following error when it runs:

assert failed: xQueueGiveFromISR queue.c:1311 (!( ( pxQueue->pcHead == ((void *)0) ) && ( pxQueue->u.xSemaphore.xMutexHolder != ((void *)0) ) ))

Lzw655, 2025-04-10:
Sorry, I made a mistake. Please replace `xSemaphoreCreateMutex()` in L756 with `xSemaphoreCreateBinary()`.

joelccrouch, 2025-04-10:
In order to use readPoints in my sketch, do I need to extern lcd_tp_mux and wrap xSemaphoreTake and xSemaphoreGive around readPoints?

joelccrouch, 2025-04-10:
I don't get the assert failed anymore, but program freezes at "Create UI" serial.print statement

Serial.println(F("Create UI"));
    /* Lock the mutex due to the LVGL APIs are not thread-safe */
    lvgl_port_lock(-1);

    /* Initialize LVGL UI generated by Squareline */
    ui_init();

......

joelccrouch, 2025-04-23:
Yes, I am using an RGB display.  The SPI conflict occurs, because I'm trying to access the touch panel in my sketch at the same time that your code in lvgl_port_v8.cpp is trying to access it.  I removed all of your changes to the lvgl_port_v8.cpp file except the semaphore wrap around the touch panel access. I then used the same semaphore wrap around my touch panel access in my sketch.  That seems to have solved my issue.  I think you should create similar functions to lvgl_port_lock(-1) and lvgl_port_unlock() but for spi.  Like spi_port_lock(-1) and spi_port_unlock() for example.

Lzw655, 2025-04-24:
Oh, I see. In this case, I think you can directly use `lvgl_port_lock(-1)` to avoid LVGL accessing the SPI bus.