esp-arduino-libs/ESP32_Display_Panel #183: Black screen on Waveshare board 5" when enable sdcard

Kanzll, 2025-04-09:
Hello,

I don't know if the problem is with board or with display panel drivers.
Everything is working well, but when I add support of sdcard using the waveshare example, the screen goes out, I see an instruction which goes off the blacklight, It's not normal, but I want to confirm with you the problem is a board design.
I use platformIO and Arduino.
here are my files:
platformio.ini
```
[env:ESP-LCD]
platform = espressif32
board = ESP-LCD
framework = arduino
#board_build.partitions = partitions.csv

platform_packages =
        platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.3
        platformio/framework-arduinoespressif32-libs @ https://github.com/esp-arduino-libs/arduino-esp32-sdk.git#high_perf/v3.0.3

        
upload_speed = 921600
monitor_speed = 115200

#build_flags =
#        -DARDUINO_USB_CDC_ON_BOOT=0
#        -DCORE_DEBUG_LEVEL=5

build_flags =
        -DBOARD_HAS_PSRAM
        -DLV_CONF_INCLUDE_SIMPLE
        -DDISABLE_ALL_LIBRARY_WARNINGS
        -DARDUINO_USB_CDC_ON_BOOT=1
        -DCORE_DEBUG_LEVEL=1
        -DLV_LVGL_H_INCLUDE_SIMPLE
        -I src

lib_deps =
        https://github.com/esp-arduino-libs/ESP32_Display_Panel.git#99b2103
        https://github.com/esp-arduino-libs/ESP32_IO_Expander.git#v0.1.0
        https://github.com/lvgl/lvgl.git#release/v8.3
        Links2004/WebSockets@^2.3.6
        bblanchon/ArduinoJson

```
main.cpp

```
#include <Arduino.h>
#include <ESP_Panel_Library.h>
#include "lvgl_port_v8.h"
#include <ui.h>
#include "waveshare_sd_card.h"

#define USE_SERIAL Serial

//SDCard 
void setup_sd_card() {
//Initialize IO expander
ESP_IOExpander *expander = new ESP_IOExpander_CH422G((i2c_port_t)I2C_MASTER_NUM, ESP_IO_EXPANDER_I2C_CH422G_ADDRESS, I2C_MASTER_SCL_IO, I2C_MASTER_SDA_IO);
expander->init();
expander->begin();
expander->multiPinMode(TP_RST | LCD_BL | LCD_RST | SD_CS | USB_SEL, OUTPUT);
expander->multiDigitalWrite(TP_RST | LCD_BL | LCD_RST, HIGH);
// Use extended GPIO for SD card
expander->digitalWrite(SD_CS, LOW);
// Turn off backlight
expander->digitalWrite(LCD_BL, LOW);
// When USB_SEL is HIGH, it enables FSUSB42UMX chip and gpio19, gpio20 wired CAN_TX CAN_RX, and then don't use USB Function 
expander->digitalWrite(USB_SEL, LOW); 
// Initialize SPI

SPI.setHwCs(false);
SPI.begin(SD_CLK, SD_MISO, SD_MOSI, SD_SS);
if (!SD.begin()) {
    Serial.println("Card Mount Failed"); // SD card mounting failed
    return;
  }
  uint8_t cardType = SD.cardType();

  if (cardType == CARD_NONE) {
      Serial.println("No SD card attached"); // No SD card connected
      return;
  }

  Serial.print("SD Card Type: "); // SD card type
  if (cardType == CARD_MMC) {
      Serial.println("MMC");
  } else if (cardType == CARD_SD) {
      Serial.println("SDSC");
  } else if (cardType == CARD_SDHC) {
      Serial.println("SDHC");
  } else {
      Serial.println("UNKNOWN"); // Unknown Type
  }

  uint64_t cardSize = SD.cardSize() / (1024 * 1024);
  Serial.printf("SD Card Size: %lluMB\n", cardSize); // SD card size
} 

void setup()
{
    USE_SERIAL.begin(115200);

    ESP_Panel *panel = new ESP_Panel();
    panel->init();

#if LVGL_PORT_AVOID_TEAR
    // When avoid tearing function is enabled, configure the RGB bus according to the LVGL configuration
    ESP_PanelBus_RGB *rgb_bus = static_cast<ESP_PanelBus_RGB *>(panel->getLcd()->getBus());
    rgb_bus->configRgbFrameBufferNumber(LVGL_PORT_DISP_BUFFER_NUM);
    rgb_bus->configRgbBounceBufferSize(LVGL_PORT_RGB_BOUNCE_BUFFER_SIZE);
#endif
    panel->begin();
    lvgl_port_init(panel->getLcd(), panel->getTouch());

    /* Lock the mutex due to the LVGL APIs are not thread-safe */
    lvgl_port_lock(-1);
    //
    ui_init();
    /* Release the mutex */
    lvgl_port_unlock();

    
    //CD_CARD
    setup_sd_card();

    // Testing file system functionality
    listDir(SD, "/", 0);

}

void loop()
{
  lv_timer_handler();
  delay(5);
 
  lv_task_handler();
}
```
Do you have any leads please?

Thank you

Lzw655, 2025-04-09:
Hi @Kanzll,

Sorry, I don't know much about this board. @H-sw123 Do you have any opinions on this question?

Lzw655, 2025-04-10:
According to the schematic information, when `LCD_BL` is set to high level, the backlight is turned on. Have you tried commenting out `expander->digitalWrite(LCD_BL, LOW);`?

Kanzll, 2025-04-10:
Yes, the screen colors deteriorate, yellow turns pink ..

Lzw655, 2025-04-10:
#168 This issue's phenomenon is very similar to yours. Please try the solution in it.

sabinhagen, 2025-04-11:
Kanzll,

I noticed that you and I are using the same exact board.  I see that you are using much older versions of the libraries than I.  Have you found that these older libraries are necessary to run the Waveshare demos?

Perhaps you and I can help each other, although I suspect you are a bit further ahead than I at getting the board and environment setup with PlatformIO.  I'm desperate to get the screen doing anything other than the ColorBarTest!

If you are willing to help a guy who will at least attempt to give you a hand as well, send me a note.  I'm at Sabinhagen@gmail.com.

ReacherS3, 2025-04-16:
@Kanzll 

Hope this helps if not I will try to assist you.

[https://github.com/ReacherS3/ESP32-S3-5inch-Display-Development-Board-32-Bit-LX7-Dual-Core-Processor-Up-To-240MHz-Frequency](url)

sabinhagen, 2025-04-18:
Hello @Kanzll,  yes,  @Lzw655 helped me properly update a missed config for I2C, and I finally got the the folder locations figured out for making the demos work with PlatformIO.

Thanks for the follow- up.  I'm nearly done with my project.  I'm not using the touchscreen,  but did change some of the settings in the lvgl_v8_port.h file and now I have a nearly completed instrument cluster project!  And LVGL is showing 62fps and ~10% CPU, so I don't think I'm going to need to worry about writing my remaining code for the other core.   

Getting the libraries configured,  AND getting the folder locations for the configs figured out was definitely the key!

Kanzll, 2025-04-22:
Hello @sabinhagen,
Very happy that you were able to achieve your goal.
Could you share your platformIO.ini and lvgl_v8_port.h to compare with mine please ?
Thank you

sabinhagen, 2025-04-22:
Sure thing!  My PlatformIO.ini has slowly been having more added as I read things or look at different examples.   In terms of LVGL performance, I didn't find any difference in most settings, including the memory allocation settings.  But the most significant was in the lvgl_v8_port.h file, I changed LVGL_PORT_AVOID_TEARING_MODE from 2 to 3, and that allowed the fps to go from 6-7 to 33.  After that change, I reduced LV_DISP_DEF_REFR_PERIOD in the lv_conf.h file from 30 (msec) to 16, and that bumped up my fps to 62.  I figured that's much faster than I need so didn't try to maximize it any further.

platformio.ini
`[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
board_build.mcu = esp32s3
board_build.arduino.memory_type = qio_opi
board_build.flash_mode = qio
board_build.psram_type = opi
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216
board_build.f_flash = 80000000L
board_build.f_cpu = 240000000L
board_build.extra_flags = 
	-DBOARD_HAS_PSRAM
	-DARDUINO_USB_CDC_ON_BOOT=1
	-DARDUINO_USB_MODE=1
board_build.partitions = partitions.csv	
build_flags =
;Arduino-related
   -DCORE_DEBUG_LEVEL=1 ; set to '5' for full debug output, '0' for none
;LVGL-related
   -DLV_CONF_INCLUDE_SIMPLE
   -DLV_LVGL_H_INCLUDE_SIMPLE
; Include Path:
   -I src 
; others:
   -DDISABLE_ALL_LIBRARY_WARNINGS    ; Disable all library warnings           
upload_speed = 	921600
monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
lib_ldf_mode = chain+
lib_deps = `

lvgl_v8_port.h
`[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
board_build.mcu = esp32s3
board_build.arduino.memory_type = qio_opi
board_build.flash_mode = qio
board_build.psram_type = opi
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216
board_build.f_flash = 80000000L
board_build.f_cpu = 240000000L
board_build.extra_flags = 
	-DBOARD_HAS_PSRAM
	-DARDUINO_USB_CDC_ON_BOOT=1
	-DARDUINO_USB_MODE=1
board_build.partitions = partitions.csv	
build_flags =
;Arduino-related
   -DCORE_DEBUG_LEVEL=1 ; set to '5' for full debug output, '0' for none
;LVGL-related
   -DLV_CONF_INCLUDE_SIMPLE
   -DLV_LVGL_H_INCLUDE_SIMPLE
; Include Path:
   -I src 
; others:
   -DDISABLE_ALL_LIBRARY_WARNINGS    ; Disable all library warnings           
upload_speed = 	921600
monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
lib_ldf_mode = chain+
[... 765 lines cut ...]
#define LV_USE_DEMO_BENCHMARK 1
#if LV_USE_DEMO_BENCHMARK
/*Use RGB565A8 images with 16 bit color depth instead of ARGB8565*/
#define LV_DEMO_BENCHMARK_RGB565A8 0
#endif

/*Stress test for LVGL*/
#define LV_USE_DEMO_STRESS 0

/*Music player demo*/
#define LV_USE_DEMO_MUSIC 1
#if LV_USE_DEMO_MUSIC
    #define LV_DEMO_MUSIC_SQUARE    0
    #define LV_DEMO_MUSIC_LANDSCAPE 0
    #define LV_DEMO_MUSIC_ROUND     0
    #define LV_DEMO_MUSIC_LARGE     0
    #define LV_DEMO_MUSIC_AUTO_PLAY 1
#endif

/*--END OF LV_CONF_H--*/

#endif /*LV_CONF_H*/

#endif /*End of "Content enable"*/
`

Kanzll, 2025-04-25:
Hello and Thanks @sabinhagen for your code,

I get many errors when compile my project using your platformIO file and using last version of Library, I think is relating with my setup, I use PlatformIO on Mac silicon.

I'll close this case because the initial problem (sdcard) was resolved, and I'll open a new case for compiling issue.

Again thank you all for your help.

closed 2025-04-25 by Kanzll