esp-arduino-libs/ESP32_Display_Panel #186: How to use in Micropython?

h3clikejava, 2025-04-18:
只看到介绍说支持Micropython，没看到文档呀？ *.py  文件在哪里找？

tsteinruecken, 2025-04-18:
You beat me by a few hours, I wanted to open the same issue :).

I’ve already tried to add esp32_display_panel as a dependency to micropython (ports/esp32) - which seems to work, however it does not show up in menuconfig. I’ve also tried to add it to ports/esp32/main_es32s3 - where menuconfig does not work at all.

Then I tried to build esp32_display_panel, esp-lib-utils and esp32_io_expander via USER_C_MODULES, which compiles but board.init() fails as there seems to be no default board configuration. Just adding my own BoardConfig to esp_panel_mp_board.cpp does also not work (board.init() works, but board.begin() crashes as somehow deep inside the esp_panel_board.cpp lcd_device->getBus() returns null and gets dereferenced).
I also tried to use the board configuration from the board_dynamic_config but ended up with missing macros (ILI9341 is undeclared and cannot be used to define ESP_PANEL_BOARD_LCD_CONTROLLER).

Without at least a basic idea how it’s supposed to work I guess I'm stuck too.

h3clikejava, 2025-04-19:
After several hours of struggling, I'm now ready to give up on MicroPython and switch to ESP-IDF.

closed 2025-04-28 by Lzw655