esp-arduino-libs/ESP32_Display_Panel #194: Support the OTA7290B controller ?

jerryliugithub, 2025-05-06:
I will use the ESP32-P4's MIPI DSI interface to drive a 7" TFT LCD with a resolution of 280x1424. The display driver IC is OTA7290B. Could you provide support for this setup?

Lzw655, 2025-05-07:
Hi @jerryliugithub,

Of course, we did drive the OTA7290 (600x1024) screen before, but we don't have this hardware now. However, I uploaded the driver to the branch [feat/add_lcd_ota7290](https://github.com/esp-arduino-libs/ESP32_Display_Panel/tree/feat/add_lcd_ota7290). Could you please help me test it? Thank you.

jerryliugithub, 2025-05-07:
Great! Let me put the hardware together first and run some tests — I’ll get back to you as soon as I have any results.