esp-arduino-libs/ESP32_Display_Panel #199: ESP32 P4 lcd_spi 示例 报告invalid sclk speed 错误，是什么原因？

linghuye, 2025-05-14:
正点原子ESP32 P4小系统板，接ST7789驱动320*240屏幕，用examples\arduino\drivers\lcd\lcd_spi做测试，
SPI屏幕频率EXAMPLE_LCD_SPI_FREQ_HZ，设置超过20M就报invalid sclk speed，这是啥原因？
```
E (122) spi_master: spi_bus_add_device(426): invalid sclk speed
E (123) lcd_panel.io.spi: esp_lcd_new_panel_io_spi(92): adding spi device to bus failed
[E][Panel][esp_panel_bus_spi.cpp:0321](begin): create panel IO failed [ESP_ERR_INVALID_ARG]
[E][Panel][esp_panel_lcd.cpp:1028](processDeviceOnInit): Bus begin failed
[E][Panel][esp_panel_lcd_st7789.cpp:0049](init): Process device on init failed
[E][Panel][esp_panel_lcd.cpp:0404](begin): Init failed
```

```
#define EXAMPLE_LCD_NAME                ST7789
#define EXAMPLE_LCD_WIDTH               (240)
#define EXAMPLE_LCD_HEIGHT              (320)
#define EXAMPLE_LCD_COLOR_BITS          (16)
#define EXAMPLE_LCD_SPI_FREQ_HZ         (40 * 1000 * 1000)

#define EXAMPLE_LCD_SPI_IO_CS       (28)
#define EXAMPLE_LCD_SPI_IO_DC       (49)
#define EXAMPLE_LCD_SPI_IO_SCK      (30)
#define EXAMPLE_LCD_SPI_IO_SDA      (29)
#define EXAMPLE_LCD_SPI_IO_SDO      (-1)
#define EXAMPLE_LCD_RST_IO          (50)  
#define EXAMPLE_LCD_BL_IO           (23) 
#define EXAMPLE_LCD_BL_ON_LEVEL     (1)
```

Lzw655, 2025-05-15:
Hi @linghuye,

请问你正在使用哪个版本的 arduino-esp32 SDK?

linghuye, 2025-05-17:
用的支持P4开发板的

`https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip`

```
Platform espressif32 @ 54.3.20 (required: espressif32 @ 54.3.20)
├── framework-arduinoespressif32 @ 3.2.0 (required: https://github.com/espressif/arduino-esp32/releases/download/3.2.0/esp32-3.2.0.zip)
├── framework-arduinoespressif32-libs @ 5.4.0+sha.2f7dcd862a (required: https://github.com/espressif/esp32-arduino-lib-builder/releases/download/idf-release_v5.4/esp32-arduino-libs-idf-release_v5.4-2f7dcd86-v1.zip)
├── tool-cmake @ 3.30.2 (required: platformio/tool-cmake @ ~3.30.2)
├── tool-esptoolpy @ 4.8.9 (required: https://github.com/pioarduino/esptool/releases/download/v4.8.9/esptool.zip)
├── tool-mkfatfs @ 2.0.1 (required: platformio/tool-mkfatfs @ ~2.0.0)
├── tool-mklittlefs @ 3.2.0 (required: tasmota/tool-mklittlefs @ ^3.2.0)
├── tool-mkspiffs @ 2.230.0 (required: platformio/tool-mkspiffs @ ~2.230.0)
├── tool-ninja @ 1.9.0 (required: platformio/tool-ninja @ ^1.7.0)
├── tool-openocd-esp32 @ 2.1100.20220706 (required: platformio/tool-openocd-esp32 @ ~2.1100.0)
├── tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 (required: platformio/tool-riscv32-esp-elf-gdb @ 14.2.0+20240403)
├── tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 (required: platformio/tool-xtensa-esp-elf-gdb @ 14.2.0+20240403) 
└── toolchain-riscv32-esp @ 14.2.0+20241119 (required: platformio/toolchain-riscv32-esp @ 14.2.0+20241119)     

Libraries
├── ESP32_Display_Panel @ 1.0.3+sha.2128a6e (required: git+https://github.com/esp-arduino-libs/ESP32_Display_Panel.git)
├── ESP32_IO_Expander @ 1.1.0+sha.e79a638 (required: git+https://github.com/esp-arduino-libs/ESP32_IO_Expander.git)
└── esp-lib-utils @ 0.2.1+sha.b68e772 (required: git+https://github.com/esp-arduino-libs/esp-lib-utils.git)
```

linghuye, 2025-05-18:
感谢开源，我杀进IDF源代码里的 esp_lcd 和 spi_master: spi里bus_add_device函数，发现spi_device_interface_config_t结构体没有正确初始化clock_source成员变量，且bus_add_device里
```
SPI_CHECK((dev_config->clock_speed_hz > 0) && (dev_config->clock_speed_hz <= MIN(clock_source_hz / 2, (80 * 1000000))), "invalid sclk speed", ESP_ERR_INVALID_ARG);
```
有个除2频率限制，于是我改成
```
spi_device_interface_config_t devcfg = {
      .clock_source = SPI_CLK_SRC_SPLL,
```
OK了。奇怪，为什么我在外部直接编译入修改的 esp_lcd_panel_io_spi.c 文件，不会跟内部.a的函数产生冲突呢。

Dimision, 2026-01-29:
I'm using ESP-IDF 5.4.1 and I have been stuck on this for few days, after appending `.clock_source = SPI_CLK_SRC_SPLL,` to the end of `spi_device_interface_config_t devcfg` struct, It finally works. 

Here is the steps:
1.  Edit `your_sdk_path:\esp\sdk\v5.4.1\esp-idf\components\esp_lcd\spi\esp_lcd_panel_io_spi.c`
2.  Find function (at around line 66) `esp_lcd_new_panel_io_spi`
3.  Find struct `spi_device_interface_config_t devcfg` and append `.clock_source = SPI_CLK_SRC_SPLL,` to the end (at around line 89)