esp-arduino-libs/ESP32_Display_Panel #201: Guition JC8012P4A1

christophe195, 2025-05-21:
Hey,

I bought https://nl.aliexpress.com/item/1005008789890066.html screen, a 10.1-inch ESP32P4. I was already running an lvgl project on a previous screen but wanted to go a little bigger and move away from a fragile developer board, this seemed ideal.

Now my project is in landscape and the board refuses with the vendor supplied code to work deftly in landscape. After multiple messages they informed me that their driver does not support rotation for Arduino. So I give this library a try, starting with the rotation example.

I noticed that my board is not among the standard boards. I tried to set up my own board via custom board but it hangs. I must admit I am a bit overwhelmed by all the settings.

What should I set to get rid of this message?

`
17:34:04.166 -> [I][LvPort][esp_panel_board.cpp:0066](init): Initializing board (Custom:Custom)
17:34:04.166 -> [I][LvPort][esp_panel_board.cpp:0235](init): Board initialize success
17:34:04.166 -> [I][LvPort][esp_panel_board.cpp:0253](begin): Beginning board (Custom:Custom)
17:34:04.166 -> [I][Panel][esp_lcd_jd9365.c:0069](esp_lcd_new_panel_jd9365): version: 1.0.1
17:34:04.331 -> [I][Panel][esp_lcd_jd9365.c:0430](panel_jd9365_init): LCD ID: 93 65 04
17:34:04.462 -> [I][LvPort][esp_panel_board.cpp:0463](begin): Board begin success
17:34:04.462 -> [I][LvPort][lvgl_v8_port.cpp:0761](lvgl_port_init): Initializing LVGL display driver
`

Lzw655, 2025-05-22:
Hi @christophe195,

You can refer to the [steps](https://github.com/esp-arduino-libs/esp-lib-utils#configuration-instructions-1) here and set `ESP_UTILS_CONF_LOG_LEVEL` to `ESP_UTILS_LOG_LEVEL_WARNING` via *esp_utils_conf.h* to get rid of this part of the Log information.

Lzw655, 2025-05-22:
BTW, *simple_rotation* currently does not support rotating the MIPI LCD. Moreover, there is currently no example to implement the dynamic rotation of the MIPI LCD (i.e., controlling arbitrary rotation during runtime). If you can accept static rotation (i.e., rotating only once during power-on), please use the example [simple_port](https://github.com/esp-arduino-libs/ESP32_Display_Panel/tree/master/examples/arduino/gui/lvgl_v8/simple_port) and set `LVGL_PORT_AVOID_TEARING_MODE` in *lvgl_v8_port.h* to `3`, and finally set `LVGL_PORT_ROTATION_DEGREE` to adjust the rotation angle.

christophe195, 2025-05-22:
@Lzw655 

Thanks for your information, the problem is that the screen keeps black.

Good to know that simple_rotation not works, i only need to change the screen 1 time so te lvgl_port_rotation_degree needs to be good for me.

Lzw655, 2025-05-23:
@christophe195 

You can first check if the screen backlight is turned on. If not, you need to check if the relevant pins are normal. If it is turned on but there is no display, please refer to the above answer to set `ESP_UTILS_CONF_LOG_LEVEL` to `ESP_UTILS_LOG_LEVEL_DEBUG`, and then send me the complete LOG information. I can help you analyze the problem.

christophe195, 2025-05-23:
@Lzw655 

Thanks for the backlight check, it was off. After changing the io it is on now but the screen is black (with backlight on now)

this is my console:
`10:38:53.752 -> [D][LvPort][esp_panel_lcd.cpp:0609](drawBitmap): (@0x4ff1c95c) Exit
10:38:54.048 -> [D][LvPort][esp_panel_lcd.cpp:0540](drawBitmap): (@0x4ff1c95c) Enter
10:38:54.048 -> [D][LvPort][esp_panel_lcd.cpp:0544](drawBitmap): Param: x_start(1204), y_start(753), width(76), height(47), color_data(@0x4ff2d8b0), timeout_ms(0)
10:38:54.048 -> [D][LvPort][esp_panel_lcd.cpp:0609](drawBitmap): (@0x4ff1c95c) Exit
10:38:54.344 -> [D][LvPort][esp_panel_lcd.cpp:0540](drawBitmap): (@0x4ff1c95c) Enter
10:38:54.344 -> [D][LvPort][esp_panel_lcd.cpp:0544](drawBitmap): Param: x_start(1204), y_start(753), width(76), height(47), color_data(@0x4ff208ac), timeout_ms(0)
10:38:54.344 -> [D][LvPort][esp_panel_lcd.cpp:0609](drawBitmap): (@0x4ff1c95c) Exit
10:38:54.641 -> [D][LvPort][esp_panel_lcd.cpp:0540](drawBitmap): (@0x4ff1c95c) Enter
10:38:54.641 -> [D][LvPort][esp_panel_lcd.cpp:0544](drawBitmap): Param: x_start(1205), y_start(753), width(75), height(47), color_data(@0x4ff2d8b0), timeout_ms(0)
10:38:54.674 -> [D][LvPort][esp_panel_lcd.cpp:0609](drawBitmap): (@0x4ff1c95c) Exit
10:38:54.939 -> [D][LvPort][esp_panel_lcd.cpp:0540](drawBitmap): (@0x4ff1c95c) Enter
10:38:54.939 -> [D][LvPort][esp_panel_lcd.cpp:0544](drawBitmap): Param: x_start(1205), y_start(753), width(75), height(47), color_data(@0x4ff208ac), timeout_ms(0)
10:38:54.971 -> [D][LvPort][esp_panel_lcd.cpp:0609](drawBitmap): (@0x4ff1c95c) Exit
10:38:55.235 -> [D][LvPort][esp_panel_lcd.cpp:0540](drawBitmap): (@0x4ff1c95c) Enter
10:38:55.270 -> [D][LvPort][esp_panel_lcd.cpp:0544](drawBitmap): Param: x_start(1205), y_start(753), width(75), height(47), color_data(@0x4ff2d8b0), timeout_ms(0)
10:38:55.270 -> [D][LvPort][esp_panel_lcd.cpp:0609](drawBitmap): (@0x4ff1c95c) Exit
10:38:55.532 -> [D][LvPort][esp_panel_lcd.cpp:0540](drawBitmap): (@0x4ff1c95c) Enter
10:38:55.565 -> [D][LvPort][esp_panel_lcd.cpp:0544](drawBitmap): Param: x_start(1208), y_start(753), width(72), height(47), color_data(@0x4ff208ac), timeout_ms(0)
10:38:55.565 -> [D][LvPort][esp_panel_lcd.cpp:0609](drawBitmap): (@0x4ff1c95c) Exit`

christophe195, 2025-05-24:
i did found some settings in the orignal demo files (that i can not rotate)

now i have a white screen with black lines on it and 2 purple lines with flicker, i think the purple is from the demo file i try to open "lv_demo_widgets()". I need to say that one spot on the screen gets hot (not burning).

New console lines: https://pastebin.com/uy1xqAP4

settings i found:
`/*
 * SPDX-FileCopyrightText: 2024 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#pragma once

#include <stdint.h>
#include "soc/soc_caps.h"

#if SOC_MIPI_DSI_SUPPORTED
#include "esp_lcd_panel_vendor.h"
#include "esp_lcd_mipi_dsi.h"

#ifdef __cplusplus
extern "C" {
#endif

/**
 * @brief LCD panel initialization commands.
 *
 */
typedef struct {
    int cmd;                /*<! The specific LCD command */
    const void *data;       /*<! Buffer that holds the command specific data */
    size_t data_bytes;      /*<! Size of `data` in memory, in bytes */
    unsigned int delay_ms;  /*<! Delay in milliseconds after this command */
} jd9365_lcd_init_cmd_t;

/**
 * @brief LCD panel vendor configuration.
 *
 * @note  This structure needs to be passed to the `vendor_config` field in `esp_lcd_panel_dev_config_t`.
 *
 */
typedef struct {
    const jd9365_lcd_init_cmd_t *init_cmds;         /*!< Pointer to initialization commands array. Set to NULL if using default commands.
                                                     *   The array should be declared as `static const` and positioned outside the function.
                                                     *   Please refer to `vendor_specific_init_default` in source file.
                                                     */
    uint16_t init_cmds_size;                        /*<! Number of commands in above array */
    struct {
        esp_lcd_dsi_bus_handle_t dsi_bus;               /*!< MIPI-DSI bus configuration */
        const esp_lcd_dpi_panel_config_t *dpi_config;   /*!< MIPI-DPI panel configuration */
        uint8_t  lane_num;                              /*!< Number of MIPI-DSI lanes */
    } mipi_config;
} jd9365_vendor_config_t;

/**
 * @brief Create LCD panel for model JD9365
 *
 * @note  Vendor specific initialization can be different between manufacturers, should consult the LCD supplier for initialization sequence code.
 *
 * @param[in]  io LCD panel IO handle
 * @param[in]  panel_dev_config General panel device configuration
 * @param[out] ret_panel Returned LCD panel handle
 * @return
 *      - ESP_ERR_INVALID_ARG   if parameter is invalid
 *      - ESP_OK                on success
 *      - Otherwise             on fail
 */
esp_err_t esp_lcd_new_panel_jd9365(const esp_lcd_panel_io_handle_t io, const esp_lcd_panel_dev_config_t *panel_dev_config,
                                   esp_lcd_panel_handle_t *ret_panel);

/**
 * @brief MIPI-DSI bus configuration structure
 *
 */
#define JD9365_PANEL_BUS_DSI_2CH_CONFIG()                \
    {                                                    \
        .bus_id = 0,                                     \
        .num_data_lanes = 2,                             \
        .phy_clk_src = MIPI_DSI_PHY_CLK_SRC_DEFAULT,     \
        .lane_bit_rate_mbps = 1500,                      \
    }

/**
 * @brief MIPI-DBI panel IO configuration structure
 *
 */
#define JD9365_PANEL_IO_DBI_CONFIG()  \
    {                                 \
        .virtual_channel = 0,         \
        .lcd_cmd_bits = 8,            \
        .lcd_param_bits = 8,          \
    }

/**
 * @brief MIPI DPI configuration structure
 *
 * @note  refresh_rate = (dpi_clock_freq_mhz * 1000000) / (h_res + hsync_pulse_width + hsync_back_porch + hsync_front_porch)
 *                                                      / (v_res + vsync_pulse_width + vsync_back_porch + vsync_front_porch)
 *
 * @param[in] px_format Pixel format of the panel
 *
 */
#define JD9365_800_1280_PANEL_60HZ_DPI_CONFIG(px_format) \
    {							\
    	.virtual_channel = 0,                            \
        .dpi_clk_src = MIPI_DSI_DPI_CLK_SRC_DEFAULT,     \
        .dpi_clock_freq_mhz = 60,                        \
        .pixel_format = px_format,                       \
        .num_fbs = 1,                                    \
        .video_timing = {                                \
            .h_size = 800,                               \
            .v_size = 1280,                              \
            .hsync_pulse_width = 20,   \
            .hsync_back_porch = 20,                  \
            .hsync_front_porch = 40,                     \
            .vsync_pulse_width = 4,  				\
            .vsync_back_porch = 8,                       \
            .vsync_front_porch = 20,                     \
        },                                               \
        .flags ={					\
        	.use_dma2d = true,			\
        	},                         \
    }
#endif

#ifdef __cplusplus
}
#endif
`

Lzw655, 2025-05-26:
@christophe195 

It seems that you haven't configured `ESP_PANEL_BOARD_LCD_VENDOR_INIT_CMD`. Please modify the data here according to the initialization parameters provided by the LCD manufacturer: https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/2128a6e8c2de7231626c32a5642b1efd8fdbf76d/esp_panel_board_custom_conf.h#L293-L305

christophe195, 2025-05-26:
@Lzw655 

Wow, i have picture now. Thanks. Now only the touch. I have a GSL3680,  but ther is no GSL3680 in the list with touch drivers. Can i add them like the custom screen?

Lzw655, 2025-05-28:
Great. For GSL3680, we don't have its driver yet. Moreover, the touch can't use a universal driver like the LCD. It needs to use its own communication protocol to parse the data. If you can provide me with a copy of its data manual and a driver example, I can help you configure its driver when I have time.

christophe195, 2025-05-30:
Hey,

I must admit that I have already received much more support from you than from web store where I bought the device, thanks!

There is a zip file with all the info of this device but again some things are missing including a datasheet of the touche. Fortunately there is an arduino demo included.

I will add the file with the working example here, is that enough?

http://pan.jczn1688.com/directlink/1/HMI%20display/JC8012P4A1C_I_W_Y.zip

the example is in demo > arduino-examples > esp32p4_lvgl_v8

Lzw655, 2025-06-03:
> [@AudunKodehode](https://github.com/AudunKodehode) but you have a picture on the screen? Try to set the LVGL_PORT_ROTATION_DEGREE to 90 or 180. Does that rotate the screen?
> 
> [@Lzw655](https://github.com/Lzw655) i see the "Help Wanted" badge, how can i help?

Thank you for your concern. I'm just marking that this issue requires the help of the repository maintainer.

AudunKodehode, 2025-06-03:
@christophe195 
Yes I do have picture on the screen, but setting rotation to 0,90,180 or 270 doesn't make no difference, its still in original portait mode

AudunKodehode, 2025-06-03:
Yes did that too

AudunKodehode, 2025-06-03:
@christophe195 
Yes did that too

DejanBukovec, 2025-06-04:
@christophe195 Does now for you work everything(LCD, Touch, LVGL)? I will receive my device today or tomorrow so would like know what is state of compatibility of this lib and device...
You need to use roration 270° to get landscape position and camera on top?
Does ESP32_Display_Panel library detect touchscreen drivers? Or how you implement gsl3680 touch drivers into lib/lvgl? I see that gsl3680 touch drivers are included in Guition demo... Did you maybe use them to implement them and implement them into lvgl?

KrX3D, 2025-06-06:
hi, i would be interested if there is some working code. i got it currently working with arduino and lvgl 9.3. rotation works but its kind of "laggy"

it seems that it doesn support hardware rotation exept for mirroring (so 180 degree rotation) - panel_jd9365_mirror() in esp_lcd_jd9365.c from the demo files

christophe195, 2025-06-07:
@KrX3D : i'm working on get it working with LVGL 8, i have display but not touch. When it works i will upload a example here.
@DejanBukovec : can you please explain of you use the example from guition or from this libarary?

@Lzw655 : i have converted my configuration to a supported board file. i started from the CHSC6540 driver for the touch it uses i2c like the driver we use. but i get the errors underneath:

my new file for the board: https://pastebin.com/jsURazqW 

`
/Users/christophevandooren/Projects/MeetingRoomScreen/Arduino/libraries/ESP32_Display_Panel/src/board/esp_panel_board_default_config.cpp:212:38: error: expected primary-expression before '{' token
  212 |     .touch = BoardConfig::TouchConfig{
      |                                      ^
/Users/christophevandooren/Projects/MeetingRoomScreen/Arduino/libraries/ESP32_Display_Panel/src/board/esp_panel_board_default_config.cpp:212:38: error: expected '}' before '{' token
/Users/christophevandooren/Projects/MeetingRoomScreen/Arduino/libraries/ESP32_Display_Panel/src/board/esp_panel_board_default_config.cpp:30:52: note: to match this '{'
   30 | const BoardConfig ESP_PANEL_BOARD_DEFAULT_CONFIG = {
      |                                                    ^
/Users/christophevandooren/Projects/MeetingRoomScreen/Arduino/libraries/ESP32_Display_Panel/src/board/esp_panel_board_default_config.cpp:212:38: error: expected ',' or ';' before '{' token
  212 |     .touch = BoardConfig::TouchConfig{
      |                                      ^
/Users/christophevandooren/Projects/MeetingRoomScreen/Arduino/libraries/ESP32_Display_Panel/src/board/esp_panel_board_default_config.cpp:278:6: error: expected unqualified-id before ',' token
  278 |     },
      |      ^
/Users/christophevandooren/Projects/MeetingRoomScreen/Arduino/libraries/ESP32_Display_Panel/src/board/esp_panel_board_default_config.cpp:283:5: error: expected unqualified-id before '.' token
  283 |     .backlight = BoardConfig::BacklightConfig{
      |     ^
/Users/christophevandooren/Projects/MeetingRoomScreen/Arduino/libraries/ESP32_Display_Panel/src/board/esp_panel_board_default_config.cpp:315:6: error: expected unqualified-id before ',' token
  315 |     },
      |      ^
/Users/christophevandooren/Projects/MeetingRoomScreen/Arduino/libraries/ESP32_Display_Panel/src/board/esp_panel_board_default_config.cpp:343:5: error: expected unqualified-id before '.' token
  343 |     .stage_callbacks = {
      |     ^
/Users/christophevandooren/Projects/MeetingRoomScreen/Arduino/libraries/ESP32_Display_Panel/src/board/esp_panel_board_default_config.cpp:404:6: error: expected unqualified-id before ',' token
  404 |     },
      |      ^
/Users/christophevandooren/Projects/MeetingRoomScreen/Arduino/libraries/ESP32_Display_Panel/src/board/esp_panel_board_default_config.cpp:405:1: error: expected unqualified-id before '}' token
  405 | };
      | ^
/Users/christophevandooren/Projects/MeetingRoomScreen/Arduino/libraries/ESP32_Display_Panel/src/board/esp_panel_board_default_config.cpp:405:1: error: expected declaration before '}' token
/Users/christophevandooren/Projects/MeetingRoomScreen/Arduino/libraries/ESP32_Display_Panel/src/board/esp_panel_board.cpp: In member function 'bool esp_panel::board::Board::init()':
/Users/christophevandooren/Projects/MeetingRoomScreen/Arduino/libraries/ESP32_Display_Panel/src/board/esp_panel_board.cpp:118:50: error: 'TouchGSL3680' in namespace 'esp_panel::drivers' does not name a type
  118 |             using Touch_Class = TO_DRIVERS_CLASS(Touch, ESP_PANEL_BOARD_TOUCH_CONTROLLER);
      |                                                  ^~~~~
/Users/christophevandooren/Projects/MeetingRoomScreen/Arduino/libraries/ESP32_Display_Panel/src/board/esp_panel_board.cpp:16:49: note: in definition of macro '_TO_DRIVERS_CLASS'
   16 | #define _TO_DRIVERS_CLASS(type, name)  drivers::type ## name
      |                                                 ^~~~
/Users/christophevandooren/Projects/MeetingRoomScreen/Arduino/libraries/ESP32_Display_Panel/src/board/esp_panel_board.cpp:118:33: note: in expansion of macro 'TO_DRIVERS_CLASS'
  118 |             using Touch_Class = TO_DRIVERS_CLASS(Touch, ESP_PANEL_BOARD_TOUCH_CONTROLLER);
      |                                 ^~~~~~~~~~~~~~~~
In file included from /Users/christophevandooren/Projects/MeetingRoomScreen/Arduino/libraries/esp-lib-utils/src/esp_lib_utils.h:14,
                 from /Users/christophevandooren/Projects/MeetingRoomScreen/Arduino/libraries/ESP32_Display_Panel/src/utils/esp_panel_utils_log.h:15,
                 from /Users/christophevandooren/Projects/MeetingRoomScreen/Arduino/libraries/ESP32_Display_Panel/src/board/esp_panel_board.cpp:8:
/Users/christophevandooren/Projects/MeetingRoomScreen/Arduino/libraries/ESP32_Display_Panel/src/board/esp_panel_board.cpp:129:52: error: 'Touch_Class' was not declared in this scope; did you mean 'Bus_Class'?
  129 |                 (touch_device = utils::make_shared<Touch_Class>(touch_bus.get(), touch_config.device_config)),
      |                                                    ^~~~~~~~~~~
/Users/christophevandooren/Projects/MeetingRoomScreen/Arduino/libraries/esp-lib-utils/src/check/esp_utils_check.h:150:13: note: in definition of macro 'ESP_UTILS_CHECK_EXCEPTION_RETURN'
  150 |             x; \
      |             ^
[... cut ...]
/Users/christophevandooren/Projects/MeetingRoomScreen/Arduino/libraries/ESP32_Display_Panel/src/board/esp_panel_board.cpp:129:64: error: template argument 1 is invalid
  129 |                 (touch_device = utils::make_shared<Touch_Class>(touch_bus.get(), touch_config.device_config)),
      |                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/christophevandooren/Projects/MeetingRoomScreen/Arduino/libraries/esp-lib-utils/src/check/esp_utils_check.h:150:13: note: in definition of macro 'ESP_UTILS_CHECK_EXCEPTION_RETURN'
  150 |             x; \
`

christophe195, 2025-06-07:
what do you get when you try to rotate it?

christophe195, 2025-06-07:
yes i did get the same, and after some chat with the support of guition they only say that they did not support screen rotation on the arduino code.

DejanBukovec, 2025-06-07:
Ok but if can be done on ESP-IDF it can be done also on Arduino... Did they tell you what exactly we need to do(What initialization we need to do that screen will be rotated)?

DejanBukovec, 2025-06-07:
Ok now Im get ESP32_Display_Panel example to work.
In lvgl_v8_port.h Im set:
LVGL_PORT_AVOID_TEARING_MODE to 3
LVGL_PORT_ROTATION_DEGREE to 270

And now is everything rotated and visible as must...

```
[I][Panel][esp_panel_board.cpp:0066](init): Initializing board (Custom:Custom)
[I][Panel][esp_panel_board.cpp:0235](init): Board initialize success
[I][Panel][esp_panel_board.cpp:0253](begin): Beginning board (Custom:Custom)
[I][Panel][esp_lcd_jd9365.c:0069](esp_lcd_new_panel_jd9365): version: 1.0.1
[I][Panel][esp_lcd_jd9365.c:0430](panel_jd9365_init): LCD ID: 93 65 04
[I][Panel][esp_panel_board.cpp:0463](begin): Board begin success
Initializing LVGL
[I][LvPort][lvgl_v8_port.cpp:0748](lvgl_port_init): Avoid tearing is enabled, mode: 3, rotation: 270
[I][LvPort][lvgl_v8_port.cpp:0761](lvgl_port_init): Initializing LVGL display driver
[W][Panel][esp_panel_lcd.cpp:0796](attachRefreshFinishCallback): The previous user_data(@0x4ff13e60) is existed, will overwrite it
```

christophe195, 2025-06-07:
Great, that's where i'm now.

Sadly the driver for the touch is not in the lib, we need to work on this.

christophe195, 2025-06-07:
Strange, i do not get Serial.print to work anymore.

DejanBukovec, 2025-06-08:
This library have bug in implementation of JD9365 drivers...
When I call function:
board->getLCD()->setDisplayOnOff(false);

It turn off screen(backlight stay on). And then is not possible turn it on again... Calling function:
board->getLCD()->setDisplayOnOff(true);

Do not do anything. Also rebooting ESP do not help. Only way to get it back on is that I upload guition example...

christophe195, 2025-06-08:
Maybe you need to disable the backlight instead

DejanBukovec, 2025-06-08:
I can reproduce it... 

board->getBacklight()->off();
board->getBacklight()->on();

work as must...

If I call board->getLCD()->setDisplayOnOff(false); then it can't be turn on board->getLCD()->setDisplayOnOff(true);

Im try this few times... Only way to get LCD working again is by upload guition example which I think send some init commands to LCD which make reset...

DejanBukovec, 2025-06-10:
Ok Im figure out what I did wrong and get WORKING touch screen driver :)
In next days I will test it more and if everything will work ok I will upload it here and then someone with more knowledge can check code and use it in official library driver ...

```
Initializing board
[I][Panel][esp_panel_board.cpp:0066](init): Initializing board (Custom:Custom)
[I][Panel][esp_panel_board.cpp:0235](init): Board initialize success
[I][Panel][esp_panel_board.cpp:0253](begin): Beginning board (Custom:Custom)
[I][Panel][esp_lcd_jd9365.c:0069](esp_lcd_new_panel_jd9365): version: 1.0.1
[I][Panel][esp_lcd_jd9365.c:0430](panel_jd9365_init): LCD ID: 93 65 04
[I][Panel][esp_panel_board.cpp:0463](begin): Board begin success
Initializing LVGL
[I][LvPort][lvgl_v8_port.cpp:0748](lvgl_port_init): Avoid tearing is enabled, mode: 3, rotation: 270
[I][LvPort][lvgl_v8_port.cpp:0761](lvgl_port_init): Initializing LVGL display driver
[W][Panel][esp_panel_lcd.cpp:0796](attachRefreshFinishCallback): The previous user_data(@0x4ff13a50) is existed, will overwrite it
Creating UI
Released x=-1,y=-1 
Released x=-1,y=-1 
Released x=-1,y=-1 
Released x=-1,y=-1 
Pressed x=592,y=170 
Pressed x=547,y=219 
Pressed x=526,y=239 
Pressed x=471,y=290 
Pressed x=375,y=286 
Pressed x=321,y=289 
Pressed x=269,y=290 
Pressed x=197,y=292 
Pressed x=281,y=288 
Pressed x=393,y=290 
Pressed x=505,y=296 
Pressed x=575,y=337 
Pressed x=597,y=413 
Pressed x=616,y=470 
Pressed x=617,y=480 
Pressed x=613,y=494 
Released x=-1,y=-1 
Released x=-1,y=-1 
Released x=-1,y=-1 
Released x=-1,y=-1 
Released x=-1,y=-1 
Released x=-1,y=-1 
```

DejanBukovec, 2025-06-12:
You can't "get it working" :) You need write complete driver and modify some library files to support new driver...

[image]

[image]

[image]

And some other library files...

christophe195, 2025-06-12:
"get it working" was not the right words ;)

I did meen how you did add drivers from another project (source example)

DejanBukovec, 2025-06-12:
Because Guition have lvgl compatible drivers is "simple"... 
For base you take some existing driver in ESP32_Display_Panel library. Im use GT911 drivers and make copies of all files related to GT911 and rename copies to GSL3680. Then rename all functions in new files to gsl3680, then modify some other lib files that lib detect new driver, then you take working drivers(im use guition drivers) code compare functions, take code and make it compatible with lib and then replace in new files GT911 code with new one...
Then after dome testing and debugging you get working driver... 

Because screen is rotated for 270° by using LVGL_PORT_AVOID_TEARING_MODE=3 and LVGL_PORT_ROTATION_DEGREE(270) scaling of X and Y coordinates do not work by default as must... So someone who know this lib more and know how exactly it work can make scaling better and depend of LVGL_PORT_ROTATION_DEGREE value... Right now Im make fixed scaling for my screen and touch panel and 270°rotation...

Im test and click on button work also normaly...
```
21:47:02.198 > [Info]   (9.320, +2792)   indev_proc_press: pressed at x:1226 y:64       (in lv_indev.c line #818)
21:47:02.203 > [Info]   (9.320, +0)      lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
21:47:02.211 > [Info]   (9.324, +4)      flex_update: update 0x4ff237f8 container       (in lv_flex.c line #208)
21:47:02.450 > [Info]   (9.572, +248)    indev_proc_release: released   (in lv_indev.c line #966)
21:47:02.456 > Settings Clicked
```

Lzw655, 2025-06-17:
Hi @DejanBukovec,

Could you create a pull request with the GSL3680 driver? I can help review.

Lzw655, 2025-06-17:
> Hi [@Lzw655](https://github.com/Lzw655),
> 
> I will create pull request. Im new to lvgl and displays so hope that it is not big mess...
> 
> I need to learn how to make pull request, because I not use git a lot(Usually only for download/clone projects) ... Do I need create fork to my account, then from fork clone data to pc, make changes, commit changes to my fork and then request pull? Or is there some other way to do it? Thanks.

Yes, the process you described is completely correct. Don't worry. You can try this operation more often. It doesn't matter even if there are problems.

christophe195, 2025-06-20:
i did get it working, i will do some tests.

DejanBukovec, 2025-06-20:
Is animation slow also without my touch drivers?

AudunKodehode, 2025-06-23:
@christophe195 
I tried again and still not able to rotate the screen, I guess I must be overlooking something important, any change you could share a zip with me so I can make sure its not a HW issue?

Westcott1, 2025-06-23:
I got my panel today, and Christophe195's LVGL**9** demo code runs correctly, complete with with touch and rotation.
But how do I revert to landscape mode?
There are no rotations set anywhere in the code (that I can find).
Off-topic I know, but how to get WiFi working?
It's something to do with SDIO comms between the P4 and C6.

DejanBukovec, 2025-06-24:
For touchscreen rotation check few messages back what I change in official emo to rotate it.
For image rotation you can replace flush function with this one(Written and tested with lvgl8):

```
void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
{
  int w = area->x2 - area->x1 + 1;
  int h = area->y2 - area->y1 + 1;

  // Allocate rotated buffer (w × h pixels)
  lv_color_t *rotated = (lv_color_t *)heap_caps_malloc(w * h * sizeof(lv_color_t), MALLOC_CAP_SPIRAM | MALLOC_CAP_DMA);
  if (!rotated) {
      lv_disp_flush_ready(disp);
      return;
  }

  // Rotate 90° counterclockwise
  // dst(x, y) = src(y, w - x - 1)
  for (int y = 0; y < h; y++) {
      for (int x = 0; x < w; x++) {
          rotated[(w - x - 1) * h + y] = color_p[y * w + x];
      }
  }

  // Calculate rotated area coordinates for drawing
  int x1 = disp->ver_res - area->y2 - 1;  // new X start
  int y1 = area->x1;                              // new Y start
  int x2 = disp->ver_res - area->y1 - 1;  // new X end
  int y2 = area->x2;                              // new Y end

  lcd.lcd_draw_bitmap(x1, y1, x2 + 1, y2 + 1, (uint16_t *)rotated);

  free(rotated);
  lv_disp_flush_ready(disp);
}
```

And in setup() I use this:

```
disp_drv.hor_res = LCD_V_RES;
  disp_drv.ver_res = LCD_H_RES;
  disp_drv.physical_hor_res = LCD_V_RES;
  disp_drv.physical_ver_res = LCD_H_RES;
  disp_drv.flush_cb = my_disp_flush;
  disp_drv.draw_buf = &draw_buf;
  disp_drv.full_refresh = false;
  disp_drv.direct_mode = true;
```

Westcott1, 2025-06-24:
DejanBukovec thanks!

DejanBukovec, 2025-06-24:
@Westcott1 did you get it work with lvgl 9.0 ? For me by quick test didn't work... Also after modification of rotation I get it work with simple label with center position but not with EEZ Studio exported project(Im change from 8.X to 9.X in project settings)... I just get blank page and performance monitor in right bottom corner and RAM usage in left bottom corner...

Westcott1, 2025-06-24:
Hi DejanBukovec,
Yes, the lvgl9 example worked first-time with lvgl **9.3.0**
It just gave a white screen.
However, when I added screen bg-color and a label, they displayed OK.
BUT this example has graphics and touch "rotated by 270" in the flush and touch callbacks (bodge) - see 3 or 4 posts back.
I think the lvgl monitor stuff drives the screen directly.

christophe195, 2025-06-27:
@Westcott1 i'm also interested how we get the wifi working (mqtt and json fetching)

Westcott1, 2025-06-27:
Christophe195, have a look at this topic -
[ESP32P4/C6 WiFi init](https://github.com/espressif/arduino-esp32/issues/11404#issuecomment-3000823823)

DejanBukovec, 2025-06-27:
For me WiFi work normally(Standard WiFi library). It normally connect to WiFi network, get IP/GW/DNS from DHCP, sync clock with NTP and I can ping it from computer...
I didn't test json yet but on previous boards I use bblanchon/ArduinoJson library without issues...

Westcott1, 2025-06-27:
Thanks!
Now that my board knows that your board does connect, mine has also connected! :)>

DejanBukovec, 2025-06-28:
@christophe195 If you use drivers from my pull request then they are already rotated.
If you use Guition demo then you need change things in gsl3680_touch.cpp and esp32p4_lvgl_v8.ino and that is all :)

DejanBukovec, 2025-06-29:
Log X and Y in callback function and you will see what coordinates you get on press...

DejanBukovec, 2025-07-10:
I use:
```
    https://github.com/esp-arduino-libs/ESP32_Display_Panel.git
    https://github.com/esp-arduino-libs/ESP32_IO_Expander.git#v1.1.0
    https://github.com/esp-arduino-libs/esp-lib-utils.git#v0.2.0
    https://github.com/lvgl/lvgl.git#v8.4.0
```

Westcott1, 2025-07-10:
Hi DejanBukovec,
Yes, that's the latest ones that I'm trying.
Unfortunately `esp_lcd_mipi_dsi.h` doesn't seem to exist anywhere.
But it's a #include in esp_lcd_jd9365.h amomgst others.

DejanBukovec, 2025-07-10:
Which arduino do you use? pioarduino?
I use:

```
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
platform_packages = platformio/framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#3.2.0
```

Westcott1, 2025-07-10:
I use bare-metal Arduino IDE 2.3.5

DejanBukovec, 2025-07-11:
Im not sure but maybe this is your problem...
ESP32-P4 is fully supported in ESP-IDF v5.3, this version is used in Arduino Core v3.1.0 ...

timnijland3, 2025-07-11:
Also facing the same issue as other describe above, 90 degree SW rotate kills the performance of the JC8012P4A1 unit. I went from 40 frames per second and sub 10% CPU usage to 4 frames per second and 90% usage. Anybody found a way to hardware rotate it yet?

ajbelectronica, 2025-08-30:
Hi, could someone help me configure the module? It's a JC8012P4A, with an ESP32P4 and ESP32C6, and a 10.1-inch display. The seller didn't give me any information about these modules. I've been working with TFTs, but this has complicated things a bit for me.

KrX3D, 2025-10-17:
Hi, if anyone is interessted:

https://community.home-assistant.io/t/jc8012p4a1-guition-esp32-p4-esp32-c6/939971

JoseEnriqueFA, 2025-10-17:
JC8012P4A1  

"Can anyone confirm which GPIO pin is used to control the LCD backlight?"

Darglos13, 2025-10-17:
> > Did you finally solve the rotation issue? I'm using Pioarduino and Lvgl 9.3, and I have absolutely no idea how to rotate the display 90 or 270 degrees. Please help.
> 
> Download the zip file http://pan.jczn1688.com/directlink/1/HMI%20display/JC8012P4A1C_I_W_Y.zip You have an example how to rotate the display

I downloaded this example, it's made for the Arduino IDE. When porting the code to pioarduino, I'm encountering an i2c version conflict. No matter how hard I tried, I couldn't solve the problem.

jho59, 2025-11-17:
In chapter 10.2.27 MADCTL: Memory Access Control(36h) I see B5 PAGE/COLUMN SELECTION (MV).  That's what other controller like ST7735 use for rotation. 
And in chapter 10.2.8 Read Display Status (09h) I see '1' = Rotation (MADCTL B5='1').

So why is no hardware rotation possible?

Hema-capuf, 2025-12-10:
Hey,

I am working with a 10.1-inch 800×1280 MIPI-DSI display using the JD9365 driver on ESP32-P4 (ESP-IDF v5.5.1).
The initialization completes successfully and the panel returns a valid ID, but the screen remains black even after sending red pixel data to the display.

What is working
* Backlight turns ON
* JD9365 driver loads correctly
* Reading ID works:
    jd9365: version: 2.0.1
    jd9365: LCD ID: 93 65 04

Even when I manually draw a full-red framebuffer immediately after panel init, the screen stays black.

What I need help with
I want to confirm:
1. Are the recommended DPI timing parameters correct for this JD9365 panel?
(HSYNC/HBP/HFP, VSYNC/VBP/VFP, pixel clock)
2. Is the JD9365 driver expecting a specific porch value or PLL setting?
3. Does the JD9365 panel require any additional vendor commands after Sleep Out / Display On?
4. Is the lane bitrate (tested 1000–1500 Mbps) correct for 800×1280 RGB888 @ ~60Hz?
5. Does the driver support RGB888 mode directly, or do I need to switch to RGB565?

Hardware & Settings
* ESP32-P4
* 2 MIPI DSI lanes
* Lane bitrate tested: 1500 Mbps
* Pixel clock tested: 72 MHz and 80 MHz
* Color format: RGB888

Can anyone please help me with my problem..

JoseEnriqueFA, 2025-12-10:
My first recomendation is to try with one of the examples provied by the display manufacter.

At least is what I did replacing the demos files     

///////////////////////////////////////////////////////////////////////////////////////////
static lv_obj_t *label;
static int counter = 0;

void btn_event_cb(lv_event_t *e)
{
  counter++;
  lv_label_set_text_fmt(label, "Push: %d times", counter);
}

void create_simple_ui()
{
  lv_obj_t *btn = lv_button_create(lv_screen_active());
  lv_obj_set_size(btn, 120, 50);
  lv_obj_center(btn);
  lv_obj_add_event_cb(btn, btn_event_cb, LV_EVENT_CLICKED, NULL);

  label = lv_label_create(btn);
  lv_label_set_text(label, "Pulsado: 0 veces");
  lv_obj_center(label);
}

////////////////////////////////////////////////////////////////////////////////////////////////

void setup()
{
  Serial.begin(115200);
  Serial.println("ESP32P4 MIPI DSI LVGL");
  lcd.begin();
  touch.begin();

  lv_init();
  uint32_t buffer_size = LCD_H_RES * LCD_V_RES;

  buf = (uint32_t *)heap_caps_malloc(buffer_size, MALLOC_CAP_SPIRAM);
  buf1 =(uint32_t *)heap_caps_malloc(buffer_size, MALLOC_CAP_SPIRAM);
  assert(buf);
  assert(buf1);
  
  disp_drv = lv_display_create(LCD_H_RES, LCD_V_RES);
  lv_display_set_flush_cb(disp_drv, my_disp_flush);
  lv_display_set_buffers(disp_drv, buf, buf1, buffer_size * sizeof(uint32_t), LV_DISPLAY_RENDER_MODE_FULL);
  /*Initialize the display*/

  lv_indev_t * indev = lv_indev_create();
  lv_indev_set_type(indev, LV_INDEV_TYPE_POINTER);
  lv_indev_set_read_cb(indev, my_touchpad_read);

  create_simple_ui();    // interfaz simple

  //lv_demo_widgets(); /* 小部件示例 */
  // lv_demo_music();        /* 类似智能手机的现代音乐播放器演示 */
  // lv_demo_stress();       /* LVGL 压力测试 */
  // lv_demo_benchmark();    /* 用于测量 LVGL 性能或比较不同设置的演示 */

setup_pwm();

//set_backlight( 80);

//delay (5000);
set_backlight(20);

  
  Serial.println("setup ");
}