esp-arduino-libs/ESP32_Display_Panel #205: Waveshare ESP32-P4-NANO dev board MIPI-DSI LCD issue

RobCrown82, 2025-06-06:
The sample Arduino projects for the Waveshare ESP32-P4-NANO dev board don't properly initialize the MIPI-DSI LCD display. The display doesn't turn on (no backlight, black screen) so the code always jams in the panel_jd9365_init() function of the esp_lcd_jd9365.c file when it tries to read the LCD ID. 

The panel does initialize properly if I load the waveshare sample projects using the ESP-IDF V5.4.1 directly (through VS code). I've examined their esp_lcd_jd9365.c file and there are some i2c commands being sent to the panel in the esp_lcd_new_panel_jd9365() function that are missing in the ESP32_Display_Panel Arduino library. Those commands are in some way responsible for turning on the display. Can you please advise on how to fix this issue in your library.

I've pasted the missing i2c code below.

i2c_config_t conf = {
        .mode = I2C_MODE_MASTER,
        .sda_io_num = 7,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,
        .scl_io_num = 8,
        .scl_pullup_en = GPIO_PULLUP_ENABLE,
        .master.clk_speed = 100000,
    };

    i2c_bus_handle_t i2c0_bus = i2c_bus_create(I2C_NUM_1, &conf);
    i2c_bus_device_handle_t i2c0_device1 = i2c_bus_device_create(i2c0_bus, 0x45, 0);

    uint8_t data = 0x11;
    i2c_bus_write_bytes(i2c0_device1, 0x95, 1, &data);
    data = 0x17;
    i2c_bus_write_bytes(i2c0_device1, 0x95, 1, &data);
    data = 0x00;
    i2c_bus_write_bytes(i2c0_device1, 0x96, 1, &data);
    vTaskDelay(pdMS_TO_TICKS(100));
    data = 0xFF;
    i2c_bus_write_bytes(i2c0_device1, 0x96, 1, &data);

    i2c_bus_device_delete(&i2c0_device1);
    // i2c_bus_delete(&i2c0_bus);

    vTaskDelay(pdMS_TO_TICKS(1000));

DejanBukovec, 2025-06-09:
Hi,
JD9365 LCD Panel do not use I2C for comunication but use MIPI DSI. If you check Guition JC8012P4A1 issue thread which use 10.1 inch JD9365 LCD panel you will see example configurations for it... It also use GSL3680 touch panel which is not supported by this library...

There is no I2C configuration for JD9365. You need MIPI DSI and PWM_LED(Pin 29 in Guition example) for backlight and reset pin(Pin 23 in Guition example).

In case that your panel use GSL3680 touch panel then there is I2C configuration which use next 4 pins(In Guition example):
I2C_SDA 7
I2C_SCL 8
RST 22
INT 21
But probably your screen use different touch panel because use slower CLK speed(100kHz instead of 400kHz)...

TiAmourze, 2025-06-09:
Hello, I have the same problem with the questioner when using "**Waveshare ESP 32-P4-Nano Board MIPI-DSI LCD**". When I tried to reproduce the "board_static_config" routine in the ESP_Display_Panel library on Arduino platform, "**esp_panel_board_supported_conf**" and "**esp_panel_drivers_conf**" were correctly configured, and all the steps were carried out according to "README".

[image]

But after compiling and burning to the development board, the above debugging information appears on the serial monitor, and the whole program stops : _[D][Panel][esp_panel_lcd.cpp:0503](reset): Refresh panel(@0x48001260) reset_.

[image]

[image]

Go back to the root of the problem, trace back to the statements  ( _ESP_UTILS_CHECK_FALSE_RETURN(lcd_device->begin(), false, "LCD device begin failed");_)   in the file  (**esp_panel_board.cpp**),  then to the statements  (_ESP_UTILS_CHECK_ERROR_RETURN(esp_lcd_panel_init(refresh_panel), false, "Init panel failed");_)  in the file  (**esp_panel_lcd.cpp**),  and execute to the function (_esp_lcd_panel_init(refresh_panel)_), The whole program can't continue to execute, and the result in the figure appears.

DejanBukovec, 2025-06-11:
As Im write in previous post JD9365 do not have I2C comunication. Check documentation of JD9365 and you will see that in 220 pages of documentation there is nothing about I2C ...
Your I2C code is for touch panel GT9271. If your I2C configuration is wrong or driver do not support it fully then when you will call begin of board it will crash and your board will reboot ...
If you disable touch panel in config then your LCD(Without touchscreen) must init and work and board will not reboot...

Y1hsiaochunnn, 2025-06-12:
You need to add i2c register control, refer to this section:https://github.com/waveshareteam/Waveshare-ESP32-components/blob/db328ea6b96e7a7523ee0a8e964bb2467b486b2e/display/lcd/esp_lcd_jd9365_10_1/esp_lcd_jd9365_10_1.c#L133-L144

Y1hsiaochunnn, 2025-06-12:
> As Im write in previous post JD9365 do not have I2C comunication. Check documentation of JD9365 and you will see that in 220 pages of documentation there is nothing about I2C ... Your I2C code is [...]

I don't deny your point, but the screen control should be supported by the manufacturer, the screen has a single chip in it, you need to write registers to control reset, backlight, etc. It is also mentioned in the waveshare Wiki