esp-arduino-libs/ESP32_Display_Panel #206: Arduino Wire(i2c) library cannot be used together with this library

cheops, 2025-06-09:
I am using pioarduino release 3.2.0 (https://github.com/esp-arduino-libs/ESP32_Display_Panel/issues/147)
when using Wire from arduino together with this dirver I get an error
```
E (972) i2c: CONFLICT! driver_ng is not allowed to be used with this old driver
```

Wire I2C bus configuration
```
#include <Wire.h>
Wire.begin(SDA_PIN, SCL_PIN);
```

disabling the touch and portexpander is not enough to use this library (only for display) and also use Wire library
```
#define ESP_PANEL_DRIVERS_BUS_USE_ALL                   (0)
    #define ESP_PANEL_DRIVERS_BUS_USE_I2C               (0)
#define ESP_PANEL_DRIVERS_BUS_COMPILE_UNUSED_DRIVERS    (0)

#define ESP_PANEL_DRIVERS_TOUCH_USE_ALL                 (0)
    #define ESP_PANEL_DRIVERS_TOUCH_USE_AXS15231B       (0)
    #define ESP_PANEL_DRIVERS_TOUCH_USE_CHSC6540        (0)
    #define ESP_PANEL_DRIVERS_TOUCH_USE_CST816S         (0)
    #define ESP_PANEL_DRIVERS_TOUCH_USE_FT5x06          (0)
    #define ESP_PANEL_DRIVERS_TOUCH_USE_GT911           (0)
    #define ESP_PANEL_DRIVERS_TOUCH_USE_GT1151          (0)
    #define ESP_PANEL_DRIVERS_TOUCH_USE_SPD2010         (0)
    #define ESP_PANEL_DRIVERS_TOUCH_USE_ST1633          (0)
    #define ESP_PANEL_DRIVERS_TOUCH_USE_ST7123          (0)
    #define ESP_PANEL_DRIVERS_TOUCH_USE_STMPE610        (0)
    #define ESP_PANEL_DRIVERS_TOUCH_USE_TT21100         (0)
    #define ESP_PANEL_DRIVERS_TOUCH_USE_XPT2046         (0)
#define ESP_PANEL_DRIVERS_TOUCH_COMPILE_UNUSED_DRIVERS          (0)

#define ESP_PANEL_DRIVERS_EXPANDER_USE_ALL                      (0)
    #define ESP_PANEL_DRIVERS_EXPANDER_USE_CH422G               (0)
    #define ESP_PANEL_DRIVERS_EXPANDER_USE_HT8574               (0)
    #define ESP_PANEL_DRIVERS_EXPANDER_USE_TCA95XX_8BIT         (0)
    #define ESP_PANEL_DRIVERS_EXPANDER_USE_TCA95XX_16BIT        (0)
```

```
#define ESP_PANEL_BOARD_LCD_BUS_TYPE        (ESP_PANEL_BUS_TYPE_MIPI_DSI)
#define ESP_PANEL_BOARD_LCD_BUS_SKIP_INIT_HOST      (0)     // 0/1. Typically set to 0

#define ESP_PANEL_BOARD_USE_TOUCH               (0)
#define ESP_PANEL_BOARD_TOUCH_BUS_SKIP_INIT_HOST        (1)     // 0/1. Typically set to 0

#define ESP_PANEL_BOARD_USE_EXPANDER            (0)
#define ESP_PANEL_BOARD_EXPANDER_SKIP_INIT_HOST     (1)     // 0/1
```

wire.cpp https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/src/Wire.cpp#L36
now uses the driver_ng

cheops, 2025-06-09:
related to #147

christophe195, 2025-06-10:
Can this help? https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/docs/envs/use_with_arduino.md#how-to-avoid-i2c-re-initialization-when-using-esp32_display_panel-in-arduino-ide-eg-when-using-wire-library

cheops, 2025-06-10:
Rereading my issue, I see my explanation is not clear enough.
## case one
```C++
void setup()
{
    #include <Wire.h>
    Wire.begin(SDA_PIN, SCL_PIN);

    esp_panel::board::Board *board = new esp_panel::board::Board();
    board->init();
    assert(board->begin());
}
```

results in runtime error:
```
E (972) i2c: CONFLICT! driver_ng is not allowed to be used with this old driver
```

## case two

For this part I had enabled the I2C Driver
`#define ESP_PANEL_DRIVERS_BUS_USE_I2C               (1)`
`#define ESP_PANEL_DRIVERS_TOUCH_USE_GT911           (1)`
```C++
void setup()
{
    #include <Wire.h>
    Wire.begin(SDA_PIN, SCL_PIN);

    esp_panel::board::Board *board = new esp_panel::board::Board();
    board->init();

    /**
     * Should be called after `board->init()` and before `board->begin()`
     */
    // For I2C Touch
    static_cast<esp_panel::drivers::BusI2C *>(board->getTouch()->getBus())->configI2C_HostSkipInit();

    assert(board->begin());
}
```

results in runtime error:
```
E (972) i2c: CONFLICT! driver_ng is not allowed to be used with this old driver
```

## case three

This however runs fine (omitting the `Wire.begin()`:

```C++
void setup()
{
    #include <Wire.h>

    esp_panel::board::Board *board = new esp_panel::board::Board();
    board->init();
    assert(board->begin());
}
```

Y1hsiaochunnn, 2025-06-12:
esp-idf updated the i2c driver from the old version of i2c.h to i2c_master.h. The library is not yet fully adapted to the new driver, so it will give an error.

Developers are already adapting.

Lzw655, 2025-06-17:
Hi @cheops,

@Y1hsiaochunnn is right. The library needs to support the i2c_ng driver to resolve this issue.

cheops, 2025-06-17:
ok, from what I can see these parts need updating to i2c_ng driver
https://github.com/search?q=repo%3Aesp-arduino-libs%2FESP32_Display_Panel%20%23include%20%22driver%2Fi2c.h%22&type=code
also the ESP32_IO_Expander uses the i2c_legacy driver and needs updating
https://github.com/search?q=repo%3Aesp-arduino-libs%2FESP32_IO_Expander%20%23include%20%22driver%2Fi2c.h%22&type=code

For this project main board I do not necessary need i2c touch nor io-expander (although some of the dev boards have i2c touch)
So I will temporarily remove those blocks from the code in this project to get the main board up and running.
Let's see how that works out.

Lzw655, 2025-06-18:
> > For my project main board I do not necessary need i2c touch nor io-expander (although some of the project dev boards have i2c touch) So I will temporarily remove those blocks from the code in my project to get the project main board up and running (and the dev boards without touch). Let's see how that works out.
> 
> That works, I have the display working and Arduino Wire(i2c-ng) is also working!

Nice!

cosmicfrogpt, 2025-07-26:
Thank you for your fantastic work.

I need to use the I2C with the display (7") but after the update I am getting the same error
"i2c: CONFLICT! driver_ng is not allowed to be used with this old driver"

There's something I missed?

Thanks

Lzw655, 2025-07-30:
Hi @cosmicfrogpt,

Sorry, if you are using arduino-esp32 v3.2.0 or later, this library will indeed have this problem. I am updating the I2C-related drivers in #230.

cosmicfrogpt, 2025-07-30:
> Hi [@cosmicfrogpt](https://github.com/cosmicfrogpt),
> 
> Sorry, if you are using arduino-esp32 v3.2.0 or later, this library will indeed have this problem. I am updating the I2C-related drivers in [#230](https://github.com/esp-arduino-libs/ESP32_Display_Panel/pull/230).

Thank you for the update, I will wait 👍

lyusupov, 2025-10-07:
@Lzw655 are there any ETAs when PR https://github.com/esp-arduino-libs/ESP32_Display_Panel/pull/230 will become applied ?

LinDong88, 2025-11-10:
@Lzw655 I am unable to use this library together with any other libraries that use i2c_master.h. I attempted to make modifications but the changes were too extensive. Thank you very much for your efforts. May I ask how long it will take to complete the revision?

jlcouto, 2026-03-04:
I found a simple solution, although not the best (which is to update the driver).

Check it out here: https://github.com/esp-arduino-libs/ESP32_Display_Panel/issues/243#issuecomment-3995627482