esp-arduino-libs/ESP32_Display_Panel #217: ESPIDF : Using custom board with esp_panel_board_custom_conf.h

martinroger, 2025-06-29:
I am trying to define a custom board **out of the sdkconfig** because I have to use some vendor init commands and there is a bit of trickery on the IO expander.
I have my `main.cpp` like so : 

```
#include <stdio.h>
#include "esp_panel_board_custom_conf.h"
#include "esp_display_panel.hpp"
#include "esp_lib_utils.h"

using namespace esp_panel::drivers;
using namespace esp_panel::board;

static const char *TAG = "main";

extern "C" void app_main()
{
    Board *board = new Board();
    assert(board);

    ESP_LOGI(TAG,"Initializing board");
    ESP_UTILS_CHECK_FALSE_EXIT(board->init(),"Board init failed");
    ESP_UTILS_CHECK_FALSE_EXIT(board->begin(),"Board begin failed");
    auto lcd = board->getLCD();
    auto expander = board->getIO_Expander();
    expander->getBase()->printStatus();
    lcd->colorBarTest();

}
```

and my esp_panel_board_custom_conf.h file has `#define ESP_PANEL_BOARD_DEFAULT_USE_CUSTOM  (1)` correctly set. 

However, after compile-flash-monitor, I get an error that the default board configuration is not detected : 

```
I (664) main_task: Calling app_main()
I (664) main: Initializing board
E (674) Panel: [esp_panel_board.cpp:0055](init):
No default board configuration detected. There are three ways to provide a default configuration: 
        1. Use the `esp_panel_board_supported_conf.h` file to enable a supported board. 
        2. Use the `esp_panel_board_custom_conf.h` file to define a custom board.
        3. Use menuconfig to enable a supported board or define a custom board.
E (704) Utils: [main.cpp:0018](app_main): Board init failed
I (714) main_task: Returned from app_main()
```

Where is my mistake ?

t0menas, 2025-06-30:
Your observation that things tend to just work with arduino framework is absolutely correct, but as systems/projects become more and more complex inevitably one realizes the need for functionality that arduino framework lacks heavily. I'm glad you made the leap forward! 
For a heads up, when you'll successfully start your board and display, initially you are likely to see much worse performance from esp-idf compared to arduino framework, because how well arduino is optimized, but that's another topic, I myself am tackling this problem right now.

Now on to your issue. How about you try to add `"include/esp_panel_board_custom_conf.h"` to your `SRCS` list in `CMakeLists.txt` , from my experience including dir under `INCLUDE_DIRS` is not enough for compiler to fetch your files. your `CMakeLists.txt` should look something like this 
```
idf_component_register(
    SRCS 
        "main.cpp"
        "include/esp_panel_board_custom_conf.h"

    INCLUDE_DIRS 
        "." 
        "include"
)
```
also under INCLUDE_DIRS you listed "./include" which i cannot find anywhere to be the correct syntax, try the code above that I wrote

> I also started by unchecking "Enable to skip "esp_panel_board_*.h", but when you do that the KConfig just hides any choice other than which drivers get compiled or not.
My bad, what I meant was you probably would like to **uncheck** "Enable to skip "esp_panel_drivers_conf*.h" but **check** "Enable to skip "esp_panel_board_*.h" because drivers will be set up by your esp_panel_board_custom_conf.h but you'll have to define your board as custom in menuconfig as this will make the library search for custom config files. That is my understanding of how this works.

> or is my best way out of this to just create an empty component, plop my conf file in there and set the display library as a required sub-component ?
I don't think this is optimal solution for this case, because you'll need only a few headers, they are perfectly fine to be in your project dir as, for a lack of a better word, "definitions" 

Let me hear from You :)

martinroger, 2025-06-30:
> Your observation that things tend to just work with arduino framework is absolutely correct, but as systems/projects become more and more complex inevitably one realizes the need for functionality [...]
Aye, I can relate... especially that ultimately the objective is to be able to get more performance than with the Arduino framework !

> Now on to your issue. How about you try to add `"include/esp_panel_board_custom_conf.h"` to your `SRCS` list in `CMakeLists.txt` , from my experience including dir under `INCLUDE_DIRS` is not enough [...]
Didn't seem to change anything unfortunately.

> 
> also under INCLUDE_DIRS you listed "./include" which i cannot find anywhere to be the correct syntax, try the code above that I wrote

From my testing, both syntaxes seem to work, but you are right that yours seems more common
> My bad, what I meant was you probably would like to **uncheck** "Enable to skip "esp_panel_drivers_conf*.h" but **check** "Enable to skip "esp_panel_board_*.h" because drivers will be set up by your [...]

I actually think you were right the first time ! leaving esp_panel_drivers_conf.h checked, and esp_panel_board_*.h unchecked.

> > or is my best way out of this to just create an empty component, plop my conf file in there and set the display library as a required sub-component ?
> 
> I don't think this is optimal solution for this case, because you'll need only a few headers, they are perfectly fine to be in your project dir as, for a lack of a better word, "definitions"
> 
> Let me hear from You :)
> 
> P.S. I just found out you can use macros (pre-begin/post-begin functions) to run your vendor id's and such things before board/expander/lcd/etc. initialization, take a look at `esp_panel_board_custom_conf.h` from line 573:
> 
> ```
> ////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
> /////////////////////// Please utilize the following macros to execute any additional code if required /////////////////
> ////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
> /**
>  * @brief Pre-begin function for board initialization
>  *
>  * @param[in] p Pointer to the board object
>  * @return true on success, false on failure
>  */
> /*
> #define ESP_PANEL_BOARD_PRE_BEGIN_FUNCTION(p) \
>     {  \
>         auto board = static_cast<Board *>(p);  \
>         return true;    \
>     }
> */
> ```

Yeah, that's precisely why I wanted to stick to using the header file instead of going through the menuconfig route... my conf file is filled to the brim with those handy macros.

I think I might have cracked a way of doing this, after reading [this thread](https://esp32.com/viewtopic.php?t=33471) which recommends modifying the **project CMakeLists** instead, like so : 
```
cmake_minimum_required(VERSION 3.5)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
include_directories(${CMAKE_CURRENT_LIST_DIR}/main/include)
project(right_screen)
```
And guess what... it works ! (and is also pointing me towards some issues in my conf file, but it's easier to resolve) : 

[image]

Now waiting to see if @Lzw655 has an even better recommendation on how to do this...

t0menas, 2025-06-30:
Brilliant! Glad to see progress! 🎊 Good luck with your project :)

Lzw655, 2025-07-07:
Hi guys, 

Sorry for the late. This problem has delved into the compilation systems of Arduino and ESP-IDF. Congratulations on becoming an advanced developer!

@t0menas is right. The method of using the configuration file like *esp_panel_board_\*.h* is mainly applicable under the Arduino framework (referenced from LVGL):

1. For the Arduino framework, it requires the compiler to be able to include the directory of *esp_panel_*.h*. Since esp32-arduino version v3 and later will add the `.ino` directory to the include list by default, we can directly place the .h file in the project directory.
2. For the esp-idf framework, ESP32_Display_Panel, as a component in CMake compilation, mainly searches for header files within the scope of the standard library and the components it explicitly depends on. Placing .h files in the main directory (you can also think of main as a component that depends on all components in *components* and *managed_components* folders) cannot be directly used as an include path by ESP32_Display_Panel (unless you explicitly specify that ESP32_Display_Panel depends on main).

Therefore, in the esp-idf framework, it is best to use Kconfig to configure the custom development board. Then, you can save all custom configurations as a file through the command `idf.py save-defconfig`, similar to the file [sdkconfig.ci.custom.BOARD_ESPRESSIF_ESP32_S3_LCD_EV_BOARD_V1_5](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/test_apps/board/common/sdkconfig.ci.custom.BOARD_ESPRESSIF_ESP32_S3_LCD_EV_BOARD_V1_5). When compiling, use the command `idf.py -DSDKCONFIG_DEFAULTS="sdkconfig.defaults;sdkconfig.ci.custom.BOARD_ESPRESSIF_ESP32_S3_LCD_EV_BOARD_V1_5" set-target esp32s3` to specify which configurations to use.

@martinroger Back to the original question, indeed, it's not possible to directly configure LCD initialization parameters in Kconfig. Therefore, I reserved the [configVendorCommands()](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/2128a6e8c2de7231626c32a5642b1efd8fdbf76d/src/drivers/lcd/esp_panel_lcd.hpp#L329) interface in the LCD driver. This allows for configuration via code before `board->begin()`, similar to [the example here](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/2128a6e8c2de7231626c32a5642b1efd8fdbf76d/examples/esp_idf/lvgl_v8_port/main/main.cpp#L26-L41). However, this also reminds me to provide an example of how to use this interface. It should be difficult for users to discover this usage method.

APF-MRoger, 2025-07-07:
> [@martinroger](https://github.com/martinroger) Back to the original question, indeed, it's not possible to directly configure LCD initialization parameters in Kconfig. Therefore, I reserved the [...]

Replying from my "work" account... thanks a lot for the very useful info, definitely handy to have. I suppose that for the other PRE-x/POST-x macros of the conf file, it just has to be managed by initialising and starting the various components of the board separately rather than all at once with the `board->begin()` statement ?

Lzw655, 2025-07-07:
> Replying from my "work" account... thanks a lot for the very useful info, definitely handy to have. I suppose that for the other PRE-x/POST-x macros of the conf file, it just has to be managed by [...]

Unfortunately, these PRE-x/POST-x macros are indeed called all at once in the `board->begin()` statement now. If you need, I can add PRE-x/POST-x macros for each independent driver.

Lzw655, 2025-07-07:
> Possibly as a future extension it could be good to be able to attach callback functions to those Pre/Post points in the `board->begin()`, possibly specifying the name of the callbacks from a KConfig ? That would allow maintaining the use of menuconfig and allow users to define their own init functions for special cases... What are your thoughts?

In my understanding, the configuration of the Kconfig string attribute can only be used in the form of a string, and it seems that it cannot be combined into a function name through macros. In fact, currently, the `board->configCallback()` interface can be used to configure attaching callback functions to those Pre/Post points. I wonder if this can meet your needs?

> And I would potentially encourage adding a note for the ESP-IDF users that want to stick to using a custom conf file in the documentation, if you want I can try and make an amendment when I prepare the PR for the additional Waveshare board I have been playing with. Personally I like the cleanliness of it and that I can track versioning changes of the conf file when tweaking it, separately from the massive mess that the sdkconfig file can become (but it is a personal preference!).

That would be great! Because I am quite familiar with this process, I may not notice the hidden problems in it. Thank you very much for your contribution! 👍