esp-arduino-libs/ESP32_Display_Panel #221: ST7701 LCD driver incorrectly rejects MIPI-DSI interface due to wrong conditional compilation check

SinKy-Yan, 2025-07-03:
> I'm sorry that I'm not good at expressing myself. The following content is generated by Claude Code.

## Bug Description
  The ST7701 LCD driver (`esp_panel_lcd_st7701.cpp`) incorrectly rejects MIPI-DSI interface configuration due to a wrong conditional
   compilation check. The driver only checks for `ESP_PANEL_DRIVERS_BUS_ENABLE_RGB` but should also check for
  `ESP_PANEL_DRIVERS_BUS_ENABLE_MIPI_DSI` when using MIPI-DSI interface.

  ## Environment
  - **Platform**: ESP32-P4
  - **Library Version**: Latest (ESP32_Display_Panel)
  - **Interface**: MIPI-DSI
  - **LCD Controller**: ST7701

  ## Steps to Reproduce
  1. Configure ST7701 with MIPI-DSI interface in `esp_panel_board_custom_conf.h`:
     ```c
     #define ESP_PANEL_BOARD_LCD_CONTROLLER      ST7701
     #define ESP_PANEL_BOARD_LCD_BUS_TYPE        (ESP_PANEL_BUS_TYPE_MIPI_DSI)
  2. Enable ST7701 driver in esp_panel_drivers_conf.h:
  #define ESP_PANEL_DRIVERS_LCD_USE_ST7701    (1)
  #define ESP_PANEL_DRIVERS_BUS_USE_MIPI_DSI  (1)
  3. Compile and run the project

  Expected Behavior

  The ST7701 driver should initialize successfully with MIPI-DSI interface.

  Actual Behavior

  The driver fails with error message:
  [E][Panel][esp_panel_lcd_st7701.cpp:0073](init): MIPI-DSI is not supported

  Root Cause

  In src/drivers/lcd/esp_panel_lcd_st7701.cpp line 64-74, the conditional compilation only checks for RGB bus support:

  #if ESP_PANEL_DRIVERS_BUS_ENABLE_RGB
      // Create refresh panel
      ESP_UTILS_CHECK_ERROR_RETURN(
          esp_lcd_new_panel_st7701(
              getBus()->getControlPanelHandle(), getConfig().getDeviceFullConfig(), &refresh_panel
          ), false, "Create refresh panel failed"
      );
      ESP_UTILS_LOGD("Create refresh panel(@%p)", refresh_panel);
  #else
      ESP_UTILS_CHECK_FALSE_RETURN(false, false, "MIPI-DSI is not supported");
  #endif // ESP_PANEL_DRIVERS_BUS_ENABLE_RGB

  However, the underlying esp_lcd_new_panel_st7701() function in src/drivers/lcd/port/esp_lcd_st7701.c correctly supports both RGB
  and MIPI-DSI interfaces based on the vendor_config->flags.use_mipi_interface flag.

  Proposed Fix

  Change the conditional compilation check to include MIPI-DSI support:

  #if ESP_PANEL_DRIVERS_BUS_ENABLE_RGB || ESP_PANEL_DRIVERS_BUS_ENABLE_MIPI_DSI
      // Create refresh panel
      ESP_UTILS_CHECK_ERROR_RETURN(
          esp_lcd_new_panel_st7701(
              getBus()->getControlPanelHandle(), getConfig().getDeviceFullConfig(), &refresh_panel
          ), false, "Create refresh panel failed"
      );
      ESP_UTILS_LOGD("Create refresh panel(@%p)", refresh_panel);
  #else
      ESP_UTILS_CHECK_FALSE_RETURN(false, false, "Neither RGB nor MIPI-DSI is supported");
  #endif // ESP_PANEL_DRIVERS_BUS_ENABLE_RGB || ESP_PANEL_DRIVERS_BUS_ENABLE_MIPI_DSI

  Additional Context

  - ST7701 hardware IC supports both RGB and MIPI-DSI interfaces
  - ESP32-P4 supports MIPI-DSI interface
  - The lower-level driver implementation (esp_lcd_st7701.c) correctly handles both interfaces
  - Only the high-level wrapper has this conditional compilation issue

  Workaround

  Currently, users can work around this by adding -DESP_PANEL_DRIVERS_BUS_ENABLE_RGB=1 to build flags, but this is misleading since
  RGB interface is not actually being used.

martinroger, 2025-07-03:
You should propose a PR, sounds like you have the solution !

SinKy-Yan, 2025-07-04:
I've submitted this fix in [PR #223](https://github.com/esp-arduino-libs/ESP32_Display_Panel/pull/223)  
Fixes #221