esp-arduino-libs/ESP32_Display_Panel #229: Compiling error in LVGL example of Waveshare ESP32-S3-Touch-LCD-7

Govardhanan-btp, 2025-07-21:
Hello,

I tried compiling the LVGL example from Waveshare ESP32-S3-Touch-LCD-7's demo Code.
But I get the following error and I am helpless.

Used libraries

LVGL: 8.4.0
ESP32_Display_Panel: 0.1.4
ESP32_IO_Expander: 0.0.3

[image]

Please kindly help.

Thanks,
Gova

Lzw655, 2025-07-23:
Hi @Govardhanan-btp,

`ESP32_Display_Panel` v0.1.4 is a very early version and does not directly support Waveshare's ESP32-S3-Touch-LCD-7 development board. It is recommended that you use the latest version of `ESP32_Display_Panel`. If any problems still occur, I will work with you to solve them.

Govardhanan-btp, 2025-07-23:
Hi @Lzw655 

Thank you for your reply.
I was able to run the same example demo, downloaded from the Waveshare website.
It all runs quite good.

But when I try save anything to EEPROM using preferences library, after LVGL initialisation, I am getting the following error.

[image]

I read and write data to EEPROM, using preferences library just before LVGL initialisation in setup function and it runs smoothly. Only when I do it afterwards, like I change the value in screen and press the button called "save", the display restarts itself with the above error.

I am helpless.