esp-arduino-libs/ESP32_Display_Panel #233: Ghosting and Tearing performance issues LVGL v8

t0menas, 2025-08-25:
### BOARD: JINGCAI ESP32-4848S040C
### ENVIROMENT: ESP-IDF v5.4.1
### ESP32_Display_Panel v1.0.3, LVGL v8.4

I would appreciate any help optimizing performance in esp-idf framework.

I experience ghosting and tearing when screen changes and animations (like moving slider) occur. None of these issues in platformio (arduino framework). Here's a [video to visualize the issues](https://imgur.com/a/ldk28Iy)

Settings that I changed around with in **lvgl_v8_port.h** :

`#define CONFIG_LVGL_PORT_AVOID_TEARING_MODE     (0)`  - enabling this mode (1,2 or 3) results in huge tearing issues.

`#define LVGL_PORT_BUFFER_SIZE_HEIGHT            (40)` - the lower the better speed but no effect on ghosting
`#define LVGL_PORT_BUFFER_NUM                    (2)` - no significant impact

I did try to add this line to **main.cpp** `bus->configRGB_BounceBufferSize(480 * 20)` to no avail...

P.S. thank you LZW655 for maintaining this awesome repo and restlessly helping the growing community :)

### My files:

**main.cpp:**:

```
#include <esp_log.h>
#include <esp_display_panel.hpp>
#include "globals.hpp"
#include "lvgl_v8_port.h"
#include "lv_demos.h"

static const char *TAG = "MAIN";

Backlight *backlight = nullptr;
SemaphoreHandle_t wifi_connected_sem = nullptr;

extern "C" void app_main(void)
{
    ESP_LOGI(TAG, "Initializing board...");
    Board *board = new Board();
    assert(board);
    ESP_UTILS_CHECK_FALSE_EXIT(board->init(), "Board init failed");   // Init bus, panel, touch

    auto bus = static_cast<esp_panel::drivers::BusRGB *>(board->getLCD()->getBus());
    ESP_UTILS_CHECK_FALSE_EXIT(bus->configRGB_BounceBufferSize(480 * 40), "BUS Bounce Buffer Size setting failed");
    ESP_LOGI(TAG, "INIT finnished, beginning board:");
    ESP_UTILS_CHECK_FALSE_EXIT(board->begin(), "Board begin failed");  // Power up
    ESP_LOGI(TAG, "begin finnished, getting lcd, touch, backlight instances:");

    auto lcd = board->getLCD();
    auto touch = board->getTouch();
    backlight = board->getBacklight();

    ESP_LOGI(TAG, "Initializing LVGL...");
    ESP_UTILS_CHECK_FALSE_EXIT(lvgl_port_init(lcd, touch), "LVGL init failed");  

    ESP_LOGI(TAG, "Creating UI...");
    lvgl_port_lock(-1);   // Lock LVGL access
    
    ui_init();            // From EEZ Studio or SquareLine

    lvgl_port_unlock();  
    ESP_LOGI(TAG, "Starting app");

    if (backlight) {
        backlight->setBrightness(95);
    }
}
```

and **lvgl_v8_port.h**:
[... 154 lines cut ...]
bool lvgl_port_deinit(void);

/**
 * @brief Lock the LVGL mutex. This function should be called before calling any LVGL APIs when not in LVGL task,
 *        and the `lvgl_port_unlock()` function should be called later.
 *
 * @param timeout_ms The timeout of the mutex lock, in milliseconds. If the timeout is set to `-1`, it will wait indefinitely.
 *
 * @return true if success, otherwise false
 */
bool lvgl_port_lock(int timeout_ms);

/**
 * @brief Unlock the LVGL mutex. This function should be called after using LVGL APIs when not in LVGL task, and the
 *        `lvgl_port_lock()` function should be called before.
 *
 * @return true if success, otherwise false
 */
bool lvgl_port_unlock(void);

#ifdef __cplusplus
}
#endif

```

Lzw655, 2025-08-27:
Hi @t0menas,

I used the configuration and code you provided, and when I ran `lv_demo_widgets()`, this issue did not occur. Have you tried using this demo? If you can also reproduce the problem this way, please package the project (delete *build* and *managed_components*) and share it with me to try. Thank you!

t0menas, 2025-09-30:
Hi @Lzw655,
got back to it as soon as I could. So  I can't get `lv_demo_widgets()` to work, whatever I do compiler throws 
`undefined reference to 'lv_demo_widgets' `
but I can run other examples (lv_demo_music()) and there is still the same awful tearring issue. [Here's the project](https://drive.google.com/file/d/1LON-IvYDHxHjU0-Af8Agg86kPbleXc58/view?usp=drive_link)